open62541/open62541 #3764: Add public API to trigger secure channel renewal

clepelli, 2020-07-03:
# Description

Add public API to trigger secure channel renewal.

## Background Information / Reproduction Steps

I use a secured connection to read real time data with a need for low enough latency. In case there is too much latency, missed data in OPC server buffer is overwritten and gone forever. Data acquisition is critical during some operations, but is not in between. 

In the same time, I see that secure channel renewal may trigger a CPU peak on OPC server and have a duration around 500ms, causing a delay for my next requests.

I would like to be able to manually renew secure channel when I don't need real time.

Workaround : I use internal API and I cheat with the renewal time to trigger it when I want. 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1
 - [x] Other OPC UA SDKs used (client or server): /
 - [x] Operating system: windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-07-06:
Good idea.
Can you check that #3766 does what you need?

clepelli, 2020-07-07:
Hello,

I have not tested but I read your commit instead. What I understand is that you allow user to say "I want to renew", but if the elapsed time is not enough, nothing will happen. In my (specific ?) case, I need to be able to _actually_ trigger the renewal when I have a good time frame, which often means before the 75% are elapsed.

- The OPC server does not allow more than 10mn lifetime, so the time frame between 75% and 100% is only 2mn30s, which is less than my latency sensitive operations durations.  I want to trigger before :
`max renewal time = expiration - max critical operation duration` 
This means I want to check the need before each critical operation. This feature implies coordination between userland code and renewal operation, with access to renewal time for instance.
- what would happen with your commit is that I may ask for the renewal at 7mn, it would not happen, then I would have the choice between waiting 30s and asking again, or triggering my operation and having a failure at 7mn30s.

Also, my need is a blocking operation, so I will wrap it in a sync. (blocking) operation as you suggested.

jpfr, 2020-07-07:
The standard says that renewal can only happen once 75% of the interval has elapsed.
I would not mess with that. Otherwise we never know which remote SDK simply closes the connection on us for doing something outside the spec.

What you could do is negotiate a channel lifetime where your time-critical operations fit inside a 25% lifetime window.
Moste remote sides won't have any problem with channel lifetimes of 60minutes and more...

Merging the PR then.

clepelli, 2020-07-07:
Ok, I understand ! I don't know the standard so well, I need to connect and adapt to a single plc provider.

jpfr, 2020-07-07:
You can increase the requested SecureChannel lifetime in the client config.
The server responds with a revised lifetime that may be different.
Logging of the lifetime is added in PR  #3771 .

Closing this. Please reopen if necessary.

closed 2020-07-07 by jpfr