open62541/open62541 #3765: UA_client isn't working well. Problem with subscription.

nokem61, 2020-07-03:
## Description

Client after renew channel never reconnect. Without any monitored items there is no such problem. After renew PublishRequest i send with old Security Token id. (see in [wireshark.zip](https://github.com/open62541/open62541/files/4871448/wireshark.zip) )

## Background Information / Reproduction Steps

```
#include "pch.h"
#include <open62541/client_config_default.h>
#include <open62541/client_highlevel.h>
#include <open62541/plugin/log_stdout.h>
#include <open62541/client_subscriptions.h>

#include <signal.h>
#include <stdlib.h>

static UA_ByteString
loadFile(const char *const path)
{
  UA_ByteString fileContents = UA_STRING_NULL;

  /* Open the file */
  FILE *fp = fopen(path, "rb");
  if (!fp)
  {
    errno = 0; /* We read errno also from the tcp layer... */
    return fileContents;
  }

  /* Get the file length, allocate the data and read */
  fseek(fp, 0, SEEK_END);
  fileContents.length = (size_t)ftell(fp);
  fileContents.data = (UA_Byte *)UA_malloc(fileContents.length * sizeof(UA_Byte));
  if (fileContents.data)
  {
    fseek(fp, 0, SEEK_SET);
    size_t read = fread(fileContents.data, sizeof(UA_Byte), fileContents.length, fp);
    if (read != fileContents.length)
      UA_ByteString_clear(&fileContents);
  }
  else
  {
    fileContents.length = 0;
  }
  fclose(fp);

  return fileContents;
}
static void OnDataChanged(UA_Client * client, UA_UInt32 subscriptionId, void * subscriptionContext, UA_UInt32 monId, void * monContext, UA_DataValue * value)
{
  UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Counter: %d", *(UA_Double*)value->value.data);
}

UA_Boolean running = true;

static void stopHandler(int sign)
{
  UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, "Received Ctrl-C");
  running = 0;
}

int main(void)
{
  signal(SIGINT, stopHandler); /* catches ctrl-c */

  UA_Client *client = UA_Client_new();
  UA_ClientConfig *cc = UA_Client_getConfig(client);
  UA_ByteString certificate = loadFile("OPCUADrv.der");
  UA_ByteString privateKey = loadFile("OPCUADrv.key");
  UA_ClientConfig_setDefaultEncryption(cc, certificate, privateKey, nullptr, 0, nullptr, 0);
  cc->securityMode = UA_MESSAGESECURITYMODE_NONE;

  cc->timeout = 1000;

  cc->secureChannelLifeTime = 2000;

  UA_Client_connectUsername(client, "opc.tcp://localhost:53530/OPCUA/SimulationServer", "test1", "test");

  UA_CreateSubscriptionRequest request;
  request = UA_CreateSubscriptionRequest_default();
  request.requestedPublishingInterval = 1000;
  UA_CreateSubscriptionResponse response = UA_Client_Subscriptions_create(client, request, nullptr, nullptr, nullptr);
  UA_MonitoredItemCreateRequest request2 = UA_MonitoredItemCreateRequest_default(UA_NODEID_STRING_ALLOC(3, "Counter")); 
  UA_Client_MonitoredItems_createDataChange(client, response.subscriptionId, UA_TIMESTAMPSTORETURN_BOTH, request2, nullptr, OnDataChanged, nullptr); //when don;t create monitored item everything is fine.

  /* Endless loop */
  while (running)
  {
    /* if already connected, this will return GOOD and do nothing */
    /* if the connection is closed/errored, the connection will be reset and then reconnected */
    /* Alternatively you can also use UA_Client_getState to get the current state */
    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://NWOLOWIEC.TEL-STER.local:53530/OPCUA/SimulationServer");
    if (retval != UA_STATUSCODE_GOOD)
    {
      UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, "Not connected. Retrying to connect in 1 second");
      /* The connect may timeout after 1 second (see above) or it may fail immediately on network errors */
      /* E.g. name resolution errors or unreachable network. Thus there should be a small sleep here */
      UA_sleep_ms(1000);
      continue;
    }
    retval = UA_Client_run_iterate(client, 1000);
  };

  /* Clean up */
  UA_Client_delete(client); /* Disconnects the client internally */
  return EXIT_SUCCESS;
}
```

Used CMake options:

```bash
cmake -G "Visual Studio 15 2017 Win64" ..\.. -DCMAKE_INSTALL_PREFIX:PATH="../install_64_debug" -DCMAKE_BUILD_TYPE:STRING="Debug" -DUA_ARCHITECTURE:STRING="win32" -DUA_NAMESPACE_ZERO:STRING="FULL" -DUA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL="1" -DUA_ENABLE_ENCRYPTION_OPENSSL:BOOL="1" -DOPENSSL_INCLUDE_DIR:PATH="OpenSSL/include64" -DSSL_EAY_RELEASE:FILEPATH="../../OpenSSL/bin/x64/Release/libssl" -DLIB_EAY_DEBUG:FILEPATH="../../OpenSSL/bin/x64/Debug/libcrypto" -DSSL_EAY_DEBUG:FILEPATH="../../OpenSSL/bin/x64/Debug/libssl" -DLIB_EAY_RELEASE:FILEPATH="../../OpenSSL/bin/x64/Release/libcrypto" -DUA_ENABLE_HISTORIZING:BOOL="1" -DBUILD_SHARED_LIBS:BOOL="1" -DUA_LOGLEVEL:STRING="100" -DUA_ENABLE_DISCOVERY:BOOL="1"  -DUA_DEBUG:BOOL="1" -DCMAKE_SYSTEM_VERSION=10.0.17763.0
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1
 - [x] Other OPC UA SDKs used (client or server): Prosys OPC UA Simulation Server
 - [x] Operating system: Windows 10 
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached [logs.txt](https://github.com/open62541/open62541/files/4871349/logs.txt) maybe PROSYS logs will be useful 
[Prosys_log.txt](https://github.com/open62541/open62541/files/4871363/Prosys_log.txt)
 - [x] Wireshark network dump attached 
[wireshark.zip](https://github.com/open62541/open62541/files/4871448/wireshark.zip)
 - [x] Self-contained code example attached
 - [x] Critical issue

jpfr, 2020-07-03:
Switch to version 1.1.1.
That resolves the bug.

nokem61, 2020-07-04:
Ok. I didn't see 1.1.1 update.
Thanks.

closed 2020-07-04 by nokem61