open62541/open62541 #3768: building open62541 in STM32 with freeRTOS but without LwIP

santoshpanda1989, 2020-07-07:
- [ ] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached yes
 - [ ] Critical issue

Hi,

I am trying to create a new architecture based on FreeRTOS and Atmel WINC1500 WiFi module. But, I am a bit confused w.r.t. the architecture header file. Anyone has any experience with WINC1500 and open62541? I have attached the ua_architecture.h file based on ATWINC1500 and have used the ua_freertos.h in the common folder.
[ua_arch.txt](https://github.com/open62541/open62541/files/4883477/ua_arch.txt)

Thanks,
Santosh

cochicde, 2020-07-07:
Hi, 

I don't know about your module, but I do know about the files and architectures. 

First of all, check the readme file in the arch folder. You have some information there.

I'd recommend to copy the freeRTOSLwIP arch folder and rename it. You will need the same ua_clock.c file since it's freeRTOS (now that I think, this file should be in the common folder). The ua_architecture_functions.c file implements four functions related to the networking. Check the lwip and posix and see how this is implemented. 

I'd recommend to create one header file where all things related to the wifi module are there, similar to the ua_lwip.h one in the common header. This way, the same header could be used by other people using other OS instead of freeRTOS. I think you shoudl put there all the things in the file you attached.

And the ua_archicture.h header will be almost similar to the one in freeRTOSLwIP with the exception that intead of including ua_lwip.h you will include yours.

Then create the CMakeList.txt file similar to freeRTOSLwIP and give it a try.

mohamedalielghoula, 2020-09-29:
Hello all
I use stm32 as an OPC server, and the timestamp starts at 01:00
I want to change this time manually or is there another way to get this time from the host automatically

teccheck, 2025-05-20:
Since 1.4 the arch integration was overhauled and redefined. Currently, we are working on a freeRTOS implementation with lwip (see #7174)

closed 2025-05-20 by teccheck