open62541/open62541 #3769: onRead Callback using a pointer

viktor-morin, 2020-07-07:
- [ 2d2b98b ] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server): None
 - [x] Operating system: Linux
 - [] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

We are using a server, where we today use callback, etc:
```
UA_NodeId currentNodeId = UA_NODEID_STRING(1, "current-time-value-callback");
UA_ValueCallback callback ;
callback.onRead = beforeReadTime;
UA_Server_setVariableNode_valueCallback(server, currentNodeId, callback);

static void
beforeReadTime(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeid, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data) {
    updateCurrentTime(server);
}
```

We have alot of nodes, and to create a callback function foreach "onRead" isnt sutiable for us.
The only thing we are doing is that we point the value from a memory to the node.

```
float example = data.floatValue;
UA_Variant value;
UA_Variant_setScalar(&value, &data.floatValue, &UA_TYPES[UA_TYPES_FLOAT]);
UA_NodeId currentNodeId = UA_NODEID_STRING(1, "current-time-value-callback");
UA_Server_writeValue(server, currentNodeId, value);
```

Is there a way to just give it a pointer or similar? Or what's the best implementation?
We are not doing anything else fancy in the "onRead" callback.

ccvca, 2020-07-07:
You can provide a poitner for a node, which is passed as "void *nodeContext" using UA_Server_setNodeContext.

viktor-morin, 2020-07-07:
> You can provide a pointer for a node, which is passed as "void *nodeContext" using UA_Server_setNodeContext.
Thanks for replying!
Do you have an example I can look at? Just a glimpse :)

ccvca, 2020-07-07:
Just call UA_Server_setNodeContext before UA_Server_setVariableNode_valueCallback and the poitner will be there. I do not have a small example at the moment.

viktor-morin, 2020-07-07:
> UA_Server_setVariableNode_valueCallback

The two functions:
[https://open62541.org/doc/current/server.html?highlight=ua_server_setmethodnode_callback#method-callbacks](https://open62541.org/doc/current/server.html?highlight=ua_server_setmethodnode_callback#method-callbacks)
[https://open62541.org/doc/current/server.html?highlight=ua_server_setnodecontext#data-source-callback](https://open62541.org/doc/current/server.html?highlight=ua_server_setnodecontext#data-source-callback)

I can't find any example or usage of them. Not sure how they should be used :/
I continue look into the other issues here and see if someone has used this.

jpfr, 2020-07-07:
See the examples here: https://github.com/open62541/open62541/blob/master/examples/tutorial_server_datasource.c

viktor-morin, 2020-07-07:
> See the examples here: https://github.com/open62541/open62541/blob/master/examples/tutorial_server_datasource.c

Thanks for replying @jpfr 
I can see the usage of 

> UA_Server_setVariableNode_valueCallback

which we use currently (see first post). Other than that, I cant see what we are looking for.
We are looking for a way to reference a pointer or similar to a NodeId where the dataValue should be read/written from. The dataValue is stored in a shared memory and used from other services as well.

jpfr, 2020-07-07:
See the comment by @ccvca 

> UA_Server_setNodeContext 

The context is attached to the individual node and forwarded to the onRead callback.
So you can more easily reuse the callback method.

The context can already be set during the node creation if that is easier.

viktor-morin, 2020-07-08:
Thanks for all help! @jpfr @ccvca 
Manage to do what we want, I can now setup everything during the creation of the code.
Here is just an example on how it can be done.

```
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

typedef struct {

    int value;

} TestClass;

TestClass testClass;

static void
addIntegerValue(UA_Server *server, UA_NodeId currentNodeId) {
    UA_UInt16 value = 0;
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "IntegerValue");
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    UA_Variant_setScalar(&attr.value, &value, &UA_TYPES[UA_TYPES_INT16]);

    UA_QualifiedName currentName = UA_QUALIFIEDNAME(1, "IntegerValue");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);
    UA_Server_addVariableNode(server, currentNodeId, parentNodeId,
                              parentReferenceNodeId, currentName,
                              variableTypeNodeId, attr, NULL, NULL);
}

static void
beforeReadTime(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeid, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data) {

    UA_UInt16 intValue = *(int *)nodeContext;
    UA_Variant value;
    UA_Variant_setScalar(&value, &intValue, &UA_TYPES[UA_TYPES_INT16]);
    UA_Server_writeValue(server, UA_NODEID_NUMERIC(1, nodeid->identifier.numeric), value);
}

static void
afterWriteTime(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeId, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                "The variable was updated");
}

static void
addIntegerCallback(UA_Server *server, UA_NodeId currentNodeId, int* dataPointer) {
    UA_ValueCallback callback;
    callback.onRead = beforeReadTime;
    callback.onWrite = afterWriteTime;

    UA_Server_setNodeContext(server, currentNodeId, dataPointer);
    UA_Server_setVariableNode_valueCallback(server, currentNodeId, callback);
}

static UA_StatusCode
readCurrentTime(UA_Server *server,
                const UA_NodeId *sessionId, void *sessionContext,
                const UA_NodeId *nodeId, void *nodeContext,
                UA_Boolean sourceTimeStamp, const UA_NumericRange *range,
                UA_DataValue *dataValue) {
    UA_DateTime now = UA_DateTime_now();
    UA_Variant_setScalarCopy(&dataValue->value, &now,
                             &UA_TYPES[UA_TYPES_DATETIME]);
    dataValue->hasValue = true;
    return UA_STATUSCODE_GOOD;
}

static volatile UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    testClass.value = 42;
    int *pointer;
    pointer = &testClass.value;

    addIntegerValue(server, UA_NODEID_NUMERIC(1, 1337));
    addIntegerCallback(server, UA_NODEID_NUMERIC(1, 1337), pointer);

    UA_StatusCode retval = UA_Server_run_startup(server);

    while(running) 
    {
        UA_UInt16 timeout = UA_Server_run_iterate(server, false);
        struct timeval tv;
        tv.tv_sec = 0;
        tv.tv_usec = timeout * 200;
        select(0, NULL, NULL, NULL, &tv);

        testClass.value = testClass.value + 1;
    }

    UA_Server_run_shutdown(server);
    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

closed 2020-07-08 by viktor-morin