open62541/open62541 #3770: Minor issue with openssl and application URI verification

m-drozd, 2020-07-07:
## Description
Hi. After migrating from 1.0.1 to 1.1.1 and switching from MbedTLS to OpenSSL I've noticed this warning when connecting to a server 
`The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None`

This seems to be mostly harmless although I can't really be sure about that.

## Background Information / Reproduction Steps

I'm using a lightweight c++ wrapper for the client.
This method sets the application uri:
```c++ 
void OpcUA::Open62541Wrapper::setApplicationUri(const std::string &applicationUri)
{
	auto cfg = UA_Client_getConfig(_client);
	UA_String_clear(&cfg->clientDescription.applicationUri);
	cfg->clientDescription.applicationUri = UA_String_fromChars(applicationUri.c_str());
}
````
The URI is set before any connect functions are called. This is what I can see when debugging
[image]

Notice the '\200' at the end which should not be a part of the string (since we have explicit length of 32), but merely a random byte past the allocated memory. That byte however seems to cause an issue here:
https://github.com/open62541/open62541/blob/v1.1.1/plugins/ua_pki_default.c#L70

When comparing the URIs from configuration and certificate the length seems to be ignored and due to that '\200' the strings are  deemed not equal.

Comparison of first character:
[image]

Comparison of "last" character:
[image]

After that the '/200' byte is compared which causes the error:
[image]

Maybe I'm missing something obvious here, but I think that the last iteration should not happen since it's effectively 33rd byte being compared.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=REDUCED -DUA_NO_AMALGAMATION -DUA_ENABLE_ENCRYPTION_OPENSSL
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux 5.5.8
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-07-07:
> This seems to be mostly harmless although I can't really be sure about that.

It is not.

Most remote clients/servers will refuse the connection if the certificate uses a different URI from the ApplicationDescription sent during the OpenSecureChannel handshake.

This was always the case. But only now do we issue a warning as a convience.

m-drozd, 2020-07-08:
> It is not.
>
> Most remote clients/servers will refuse the connection if the certificate uses a different URI from the ApplicationDescription sent during the OpenSecureChannel handshake.
>
> This was [...]

I understand the purpose of the warning, what I'm trying to say is I don't understand why I'm seeing it, since from what I can tell the URI is set correctly. The connection actually works fine and according to wireshark the transmitted URI what I set it to be.

[image]

If it's correct here then the issue seems to be only with the validation, although after further digging I think that the UA_Bstrstr function is actually fine here.

mtenberge, 2020-08-28:
I see the same behavior here. What I noticed is, that the warning message is logged *for each endpoint* when I really set it to an incorrect value. When I set it to the correct value, all warnings disappear, except the one for the endpoint with SecurityPolicy None.

Sounds logical to me, since that endpoint doesn't have a certificate, so there is no way the ApplicationUri could match with it...

Shouldn't this message be suppressed for unencrypted endpoints?

masteralarm, 2020-11-17:
The "problem" seems to come from this [function](https://github.com/open62541/open62541/blob/master/src/client/ua_client_connect.c#L964) in `ua_client_connect.c`.

Question is how to fix this. Should we change the behaviour ...
- So that `verifyClientApplicationURI` filters out the check for `SecurityPolicyNone`?
- So that `verifyClientApplicationURI` better handles the return value of the pki's `verifyApplicationURI` function (only log error when StatusCode is `UA_STATUSCODE_BADCERTIFICATEURIINVALID`)?

What do you think?

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier