open62541/open62541 #3773: UA_Variant_setArray result in segmentation fault

viktor-morin, 2020-07-08:
- [ x ] open62541 Version: master
 - [ x ] Other OPC UA SDKs used (client or server): None
 - [ x ] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

I add an array using following code:
(used this as inspiration: [https://github.com/open62541/open62541/blob/master/examples/pubsub/server_pubsub_publisher_iop.c](https://github.com/open62541/open62541/blob/master/examples/pubsub/server_pubsub_publisher_iop.c)
```
addVectorValue(server, UA_NODEID_NUMERIC(1, 1338));
```

```
static void addVectorValue(UA_Server *server, UA_NodeId currentNodeId) {
    UA_UInt32 arrayDims = 0;
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.valueRank = UA_VALUERANK_ONE_DIMENSION;
    attr.arrayDimensions = &arrayDims;
    attr.arrayDimensionsSize = 1;
    UA_Variant_setArray(&attr.value, UA_Array_new(1000, &UA_TYPES[UA_TYPES_FLOAT]), 1000, &UA_TYPES[UA_TYPES_FLOAT]);

    UA_QualifiedName currentName = UA_QUALIFIEDNAME(1, "VectorValue");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);
    UA_Server_addVariableNode(server, currentNodeId, parentNodeId,
                            parentReferenceNodeId, currentName,
                            variableTypeNodeId, attr, NULL, NULL);
}
```
I then have a callback where I want to update the array:
```
static void
beforeReadVector(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeid, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data) {

    UA_UInt16 intValue = *(int *)nodeContext;
    UA_UInt32 nodeId = nodeid->identifier.numeric;
    printf("Vector pointerValue: %d \n", intValue);
    printf("Vector nodeId: %d \n", nodeId);

    UA_Variant value;
    UA_Float tempArray[1000];
    tempArray[0] = (UA_Float)intValue;

    UA_Variant_setArray(&value, tempArray, 1000, &UA_TYPES[UA_TYPES_FLOAT]);
    UA_Server_writeValue(server, UA_NODEID_NUMERIC(1, nodeid->identifier.numeric), value);
}
```

The last line gives a segementation error:

>     UA_Server_writeValue(server, UA_NODEID_NUMERIC(1, nodeid->identifier.numeric), value);

Anyone know what I'm doing wrong?
Thanks

jpfr, 2020-07-08:
Use UA_Variant_setArrayCopy.

jpfr, 2020-07-08:
Scratch that. My mistake.

Can you run the executable with valgrind and paste the output?

viktor-morin, 2020-07-09:
> Scratch that. My mistake.
> 
> Can you run the executable with valgrind and paste the output?

Thanks for replying @jpfr 
Here is the example code:

```
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

typedef struct {

    int value;

} TestClass;

TestClass testClass;

static void addVectorValue(UA_Server *server, UA_NodeId currentNodeId) {
    UA_UInt32 arrayDims = 0;
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.valueRank = UA_VALUERANK_ONE_DIMENSION;
    attr.arrayDimensions = &arrayDims;
    attr.arrayDimensionsSize = 1;
    UA_Variant_setArray(&attr.value, UA_Array_new(1000, &UA_TYPES[UA_TYPES_FLOAT]), 1000, &UA_TYPES[UA_TYPES_FLOAT]);

    UA_QualifiedName currentName = UA_QUALIFIEDNAME(1, "VectorValue");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);
    UA_Server_addVariableNode(server, currentNodeId, parentNodeId,
                            parentReferenceNodeId, currentName,
                            variableTypeNodeId, attr, NULL, NULL);
}

static void
beforeReadVector(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeid, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data) {

    UA_UInt16 intValue = *(int *)nodeContext;
    UA_UInt32 nodeId = nodeid->identifier.numeric;
    printf("Vector pointerValue: %d \n", intValue);
    printf("Vector nodeId: %d \n", nodeId);

    UA_Variant value;
    UA_Float tempArray[1000];
    tempArray[0] = (UA_Float)intValue;

    UA_Variant_setArray(&value, tempArray, 1000, &UA_TYPES[UA_TYPES_FLOAT]);
    UA_Server_writeValue(server, UA_NODEID_NUMERIC(1, nodeid->identifier.numeric), value);
}

static void
addVectorCallback(UA_Server *server, UA_NodeId currentNodeId, int* dataPointer) {
    UA_ValueCallback callback;
    callback.onRead = beforeReadVector;

    UA_Server_setNodeContext(server, currentNodeId, dataPointer);
    UA_Server_setVariableNode_valueCallback(server, currentNodeId, callback);
}

static volatile UA_Boolean running = true;
static void stopHandler(int sign) {
[... 96 lines cut ...]
==429==    by 0x511532B: UA_Server_processServiceOperations (ua_server_utils.c:286)
==429==
==429== HEAP SUMMARY:
==429==     in use at exit: 4,681,373 bytes in 29,471 blocks
==429==   total heap usage: 107,238 allocs, 77,767 frees, 408,421,876 bytes allocated
==429==
==429== 4,000 bytes in 1 blocks are definitely lost in loss record 28,183 of 28,203
==429==    at 0x4C31B25: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==429==    by 0x109242: addVectorValue (in /mnt/c/Linux/opcua_server_test/server)
==429==    by 0x10964D: main (in /mnt/c/Linux/opcua_server_test/server)
==429==
==429== LEAK SUMMARY:
==429==    definitely lost: 4,000 bytes in 1 blocks
==429==    indirectly lost: 0 bytes in 0 blocks
==429==      possibly lost: 0 bytes in 0 blocks
==429==    still reachable: 4,677,373 bytes in 29,470 blocks
==429==         suppressed: 0 bytes in 0 blocks
==429== Reachable blocks (those to which a pointer was found) are not shown.
==429== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==429==
==429== For counts of detected and suppressed errors, rerun with: -v
==429== Use --track-origins=yes to see where uninitialised values come from
==429== ERROR SUMMARY: 3 errors from 3 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
```

jpfr, 2020-07-09:
@andreasebner might that have to do with the new datasource code?

viktor-morin, 2020-07-09:
Can also verify that the issue exist with release 1.0.1
[https://github.com/open62541/open62541/releases/tag/v1.0.1](https://github.com/open62541/open62541/releases/tag/v1.0.1)

viktor-morin, 2020-07-09:
Added more properites without success

```
static void
beforeReadVector(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeid, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data) {

    UA_UInt16 intValue = *(int *)nodeContext;
    UA_UInt32 nodeId = nodeid->identifier.numeric;
    printf("Vector pointerValue: %d \n", intValue);
    printf("Vector nodeId: %d \n", nodeId);

    UA_Variant value;
    UA_UInt32 arrayDims = 1000;
    value.arrayDimensionsSize = UA_VALUERANK_ONE_DIMENSION;
    value.arrayDimensions = &arrayDims;
    value.arrayLength = 1000;

    UA_Float tempArray[1000];
    tempArray[0] = (UA_Float)intValue;

    printf("Vector[0] value: %f \n", tempArray[0]);

    UA_Variant_setArray(&value, tempArray, 1000, &UA_TYPES[UA_TYPES_FLOAT]);
    UA_Variant_setArrayCopy(&value, tempArray, 1000, &UA_TYPES[UA_TYPES_FLOAT]);
    UA_Server_writeValue(server, UA_NODEID_NUMERIC(1, nodeid->identifier.numeric), value);
}
```

jpfr, 2020-07-09:
You set the the `onWrite` callback to an uninitialized value.

This resolves the issue:
```c
    UA_ValueCallback callback;
    callback.onRead = beforeReadVector;
    callback.onWrite = NULL;
```

viktor-morin, 2020-07-09:
> You set the the `onWrite` callback to an uninitialized value.
> 
> This resolves the issue:
> 
> ```c
>     UA_ValueCallback callback;
>     callback.onRead = beforeReadVector;
>     callback.onWrite = NULL;
> ```

Thanks, adding onWrite fixed the issue!
@jpfr Just a question before i close this issue, so it's required to have both onRead and onWrite?
My onWrite function wont do anthing? There is no "default" you can use?

Full example:
```
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

typedef struct {

    int value;

} TestClass;

TestClass testClass;

static void addVectorValue(UA_Server *server, UA_NodeId currentNodeId) {
    UA_UInt32 arrayDims = 1000;
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.valueRank = UA_VALUERANK_ONE_DIMENSION;
    attr.arrayDimensions = &arrayDims;
    attr.arrayDimensionsSize = 1;
    UA_Variant_setArray(&attr.value, UA_Array_new(1000, &UA_TYPES[UA_TYPES_FLOAT]), 1000, &UA_TYPES[UA_TYPES_FLOAT]);

    UA_QualifiedName currentName = UA_QUALIFIEDNAME(1, "VectorValue");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);
    UA_Server_addVariableNode(server, currentNodeId, parentNodeId,
                            parentReferenceNodeId, currentName,
                            variableTypeNodeId, attr, NULL, NULL);
}

static void
beforeReadVector(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeid, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data) {

    UA_UInt16 intValue = *(int *)nodeContext;
    UA_UInt32 nodeId = nodeid->identifier.numeric;
    printf("Vector pointerValue: %d \n", intValue);
    printf("Vector nodeId: %d \n", nodeId);

    UA_Variant value;
    UA_Float tempArray[1000];
    tempArray[0] = (UA_Float)intValue;

    printf("Vector[0] value: %f \n", tempArray[0]);

    UA_Variant_setArray(&value, tempArray, 1000, &UA_TYPES[UA_TYPES_FLOAT]);
    UA_Server_writeValue(server, UA_NODEID_NUMERIC(1, nodeid->identifier.numeric), value);
}

static void
afterWriteVector(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeId, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                "The variable was updated");
}

static void
addVectorCallback(UA_Server *server, UA_NodeId currentNodeId, int* dataPointer) {
    UA_ValueCallback callback;
    callback.onRead = beforeReadVector;
    callback.onWrite = afterWriteVector;w

    UA_Server_setNodeContext(server, currentNodeId, dataPointer);
    UA_Server_setVariableNode_valueCallback(server, currentNodeId, callback);
}

static volatile UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    testClass.value = 42;
    int *pointer;
    pointer = &testClass.value;
    
    addVectorValue(server, UA_NODEID_NUMERIC(1, 1338));
    addVectorCallback(server, UA_NODEID_NUMERIC(1, 1338), pointer);

    UA_StatusCode retval = UA_Server_run_startup(server);

    while(running) 
    {
        UA_UInt16 timeout = UA_Server_run_iterate(server, false);
        struct timeval tv;
        tv.tv_sec = 0;
        tv.tv_usec = timeout * 200;
        select(0, NULL, NULL, NULL, &tv);

        testClass.value = testClass.value + 1;
    }

    UA_Server_run_shutdown(server);
    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

jpfr, 2020-07-09:
`onWrite` is just a notification for the user.
If you set it to NULL it won't be used at all.
In that sense, NULL is the default.

closed 2020-07-09 by jpfr

viktor-morin, 2020-07-09:
> `onWrite` is just a notification for the user.
> If you set it to NULL it won't be used at all.
> In that sense, NULL is the default.

But I cant set it to NULL?
Even if I don't use that callback, that's the reason of this issue.

viktor-morin, 2020-07-09:
Thanks!
Thought I had it like that, but see now that I hadn't given it any value at all.

```
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

typedef struct {

    int value;

} TestClass;

TestClass testClass;

static void addVectorValue(UA_Server *server, UA_NodeId currentNodeId) {
    UA_UInt32 arrayDims = 1000;
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.valueRank = UA_VALUERANK_ONE_DIMENSION;
    attr.arrayDimensions = &arrayDims;
    attr.arrayDimensionsSize = 1;
    UA_Variant_setArray(&attr.value, UA_Array_new(1000, &UA_TYPES[UA_TYPES_FLOAT]), 1000, &UA_TYPES[UA_TYPES_FLOAT]);

    UA_QualifiedName currentName = UA_QUALIFIEDNAME(1, "VectorValue");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);
    UA_Server_addVariableNode(server, currentNodeId, parentNodeId,
                            parentReferenceNodeId, currentName,
                            variableTypeNodeId, attr, NULL, NULL);
}

static void
beforeReadVector(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeid, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data) {

    UA_UInt16 intValue = *(int *)nodeContext;
    UA_UInt32 nodeId = nodeid->identifier.numeric;
    printf("Vector pointerValue: %d \n", intValue);
    printf("Vector nodeId: %d \n", nodeId);

    UA_Variant value;
    UA_Float tempArray[1000];
    tempArray[0] = (UA_Float)intValue;

    printf("Vector[0] value: %f \n", tempArray[0]);

    UA_Variant_setArray(&value, tempArray, 1000, &UA_TYPES[UA_TYPES_FLOAT]);
    UA_Server_writeValue(server, UA_NODEID_NUMERIC(1, nodeid->identifier.numeric), value);
}

static void
addVectorCallback(UA_Server *server, UA_NodeId currentNodeId, int* dataPointer) {
    UA_ValueCallback callback;
    callback.onRead = beforeReadVector;
    callback.onWrite = NULL;

    UA_Server_setNodeContext(server, currentNodeId, dataPointer);
    UA_Server_setVariableNode_valueCallback(server, currentNodeId, callback);
}

static volatile UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    testClass.value = 42;
    int *pointer;
    pointer = &testClass.value;
    
    addVectorValue(server, UA_NODEID_NUMERIC(1, 1338));
    addVectorCallback(server, UA_NODEID_NUMERIC(1, 1338), pointer);

    UA_StatusCode retval = UA_Server_run_startup(server);

    while(running) 
    {
        UA_UInt16 timeout = UA_Server_run_iterate(server, false);
        struct timeval tv;
        tv.tv_sec = 0;
        tv.tv_usec = timeout * 200;
        select(0, NULL, NULL, NULL, &tv);

        testClass.value = testClass.value + 1;
    }

    UA_Server_run_shutdown(server);
    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```