open62541/open62541 #3776: Problem on checking out v1.1.1 with submodules

jopi2016, 2020-07-09:
Hi,
I tried to checking out v1.1.1 with submodules

What I did is 
```
git clone --recurse-submodules git://github.com/PilzDE/open62541.git
git checkout 320d71b77564964379365e1390d1e372964388d7
```

This leads to:
fatal: reference is not a tree: 320d71b77564964379365e1390d1e372964388d7

Is it possible that the submodule still has pending changes?
See [link](https://stackoverflow.com/a/13425990)

If I do a 
`git checkout v1.1.1`
This leads to
error: pathspec 'v1.1.1' did not match any file(s) known to git

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2020-07-09:
320d71b77564964379365e1390d1e372964388d7 is not a submodule, its a commit on the 1.1 branch (https://github.com/open62541/open62541/commits/1.1)

Looks like git is not fetching all remote tags/commits.
It should be possible to force it by using the following commands in the cloned directory:
  git pull --all
  git checkout 320d71b77564964379365e1390d1e372964388d7 

Or if this still results in an error:
  git fetch 320d71b77564964379365e1390d1e372964388d7
  git checkout 320d71b77564964379365e1390d1e372964388d7

jopi2016, 2020-07-09:
okay, thank you for your feedback.

`git pull --all`
leads to Fetching origin, Already up to date.

ccvca, 2020-07-09:
I got it, you've not been cloned the original Repository:

>git clone --recurse-submodules git://github.com/PilzDE/open62541.git
This is the correct one: https://github.com/open62541/open62541.git

jpfr, 2020-07-09:
Good to see this resolved.
Closing.

closed 2020-07-09 by jpfr

jopi2016, 2020-07-10:
oh no. thanks. Problem between chair and keyboard.
Sorry for the confusion