open62541/open62541 #3784: UA Clietn without encryption can not connect

frax2222, 2020-07-16:
Hello,

I have built open62541.c and open62541.h from master branch 1.0 for operating system WEC7 (useing VS2008) without encryption.
I have built an OPCUA server on that basis that works with UAExpert OPCUA Client.

Now I have built an OPCUA client for WEC7.  From this client I can connect to my server.
But I can not connect to Prosys OPC UA Simulation server and I can not connect to IntegrationObjects simulation server.

There seems to be a problem with missing encryption. 
Do I have to accept that connection does not work without encryption ? 

If I need encryption, how can I add encryption ? On page https://open62541.org/doc/1.0/installing.html it is said that Ecnryption can be added via Build Options.  But I can not find it here: https://open62541.org/doc/1.0/building.html#build-options

Attached log-files of OPCUA client and screen shot of Prosys Simulation server.

Thank you !

[opcualog_myServer.txt](https://github.com/open62541/open62541/files/4929651/opcualog_myServer.txt)
[opcualog_Prosys.txt](https://github.com/open62541/open62541/files/4929652/opcualog_Prosys.txt)
[image]
[opcualog_IntegrationObjects.txt](https://github.com/open62541/open62541/files/4929653/opcualog_IntegrationObjects.txt)

jpfr, 2020-07-16:
It might be that anonymous login does not require encryption.

Password based login usually needs encryption. At least for the password itself if the channel is unencrypted.

jpfr, 2020-07-16:
Oh right. The build option for encryption is missing from the docs.
That’s a bug.

You can use cmake-GUI or ccmake to see all options.

frax2222, 2020-07-16:
Hello,

as shown in opcualog_Prosys,txt I try anonymous connection:

[2020-07-16 21:49:04] info/client 	Selected UserTokenPolicy anonymous with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None

In Prosys Users anonymous login is allowed:
[image]

frax2222, 2020-07-16:
Hello,

just an additional information: I can connect to UaCPPServer  (Demo Server V1.7.2) of Unified Automation.

So despite all standardisation this seems to be handled differently .....

jpfr, 2020-07-16:
At least integrationobjects complains about missing encryption.
Can you try with v1.1 and enabled encryption?
Error logging is better in the newer versions at well.

frax2222, 2020-07-17:
I will try but as far as I have seen I have to include mbedtls. I am not sure whether and how I can compile this stuff with VS2008. I have to use VS2008 for WEC7.

NoelGraf, 2021-05-20:
I would close the topic for now. Please reopen if required!

closed 2021-05-20 by NoelGraf