open62541/open62541 #3785: maxSessionTimeout for Client disconnection

abdo4694, 2020-07-16:
## Description

Hello :) 

I am trying to get the number of connected clients to my server using UA_Server_getStatistics

```bash
UA_ServerStatistics ss = UA_Server_getStatistics(server);
int count = ss.ss.currentSessionCount;
```
When I am connected to the client through cable the variable **count** reads 1. I need to know when a disconnection happens. I manually remove the LAN cable but this number does not go to 0.

So I thought of including a timeout of 1 second in the initialization of the server as follows:

```bash
...
UA_ServerConfig_setDefault(UA_Server_getConfig(server));
UA_Server_getConfig(server)->maxSessionTimeout = 1000.0;
...
```
But when I remove the LAN cable it takes around 5 seconds for the variable **count** to go to 0.

Any advice on that?

Thank you in advance.
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04 for the Server
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2020-07-16:
We have an internal cleanup job that is run every 5 seconds.
The counter will only be changed when the cleanup job has picked up old sessions.

However, the session will not be usable as soon as the timeout has happened.

NoelGraf, 2021-04-30:
@abdo4694 can the issue be closed?

closed 2021-04-30 by abdo4694