open62541/open62541 #3786: How to read Buildinfo data

bnadem, 2020-07-16:
## Description

 i have nodeid from Buildinfo, i want to read value? normal  type width casting
whenn i have type from BuildInfo or from Extensionobject
how to read this value, wich function ? 
thankds

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

matkonnerth, 2020-07-18:
Here a small example with reading from server, client should look in a similiar way, hope this helps:

```c
UA_Variant var;
    UA_Variant_init(&var);
    UA_StatusCode status = UA_Server_readValue(server, UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_BUILDINFO), &var);
    if(!(status==UA_STATUSCODE_GOOD))
    {
        printf("read failed");
        UA_Variant_clear(&var);
        return EXIT_FAILURE;
    }
    if(!(var.type == &UA_TYPES[UA_TYPES_BUILDINFO]))
    {
        printf("wrong datatype");
        UA_Variant_clear(&var);
        return EXIT_FAILURE;
    }
    UA_BuildInfo* buildInfo = (UA_BuildInfo*)(var.data);
    printf("Build info productName %.*s", (int)buildInfo->productName.length, buildInfo->productName.data);
    UA_Variant_clear(&var);

matkonnerth, 2020-07-24:
then this issue can be closed?

closed 2021-04-30 by NoelGraf