open62541/open62541 #3787: Client to read custom Datatype/ ExtensionObject

skubidu1324, 2020-07-17:
## Description
I want to develop a client to read a Struct/ExtensionObject from a Siemens PLC. The struct contains 16 UA_Boolean variables and is defined in root/Types/VariableTypes/BaseVariableType/
#1420 ,  #2369 and #3108 are related but havn't given a solution.  

## Background Information / Reproduction Steps
I followed the steps from `master\examples\custom_datatype`. I created the struct, UA_DataTypeMember and UA_DataType for that specific struct but only for client side because the server is given.

```
#define CustomDataStruct_binary_encoding_id        1

typedef struct {
	UA_Boolean Data01;
        ...
        UA_Boolean Data16;
} CustomDataStruct;

static UA_DataTypeMember CustomDataStruct_members[16] = {
	{
		UA_TYPENAME("Data01")		        /* .memberName */
		UA_TYPES_BOOLEAN,                       /* .memberTypeIndex, points into UA_TYPES since namespaceZero is true */
		0,                                     /* .padding */
		true,                                  /* .namespaceZero, see .memberTypeIndex */
		false,                                 /* .isArray */
		false                                  /* .isOptional */
	},
       ...
}

static const UA_DataType CustomDataStructType = {
	UA_TYPENAME("CustomDataStruct")                         
	UA_NODEID_STRING(3, "VT_\"CustomDataStruct\""), /*nodeid from root/Types/VariableTypes/BaseVariableType/*/
	sizeof(CustomDataStruct),                                        
	0, 
	UA_DATATYPEKIND_STRUCTURE, 
	false,
	false, 
	16, 
	CustomDataStruct_binary_encoding_id, 
	CustomDataStruct_members
};
```
Then added this type to the client configuration as the example shows.

```

    UA_DataType types[1];
    types[0] = CustomDataStructType;
    UA_DataTypeArray customDataTypes = {NULL, 1, types};

    UA_Client *client = UA_Client_new();
    UA_ClientConfig *cc = UA_Client_getConfig(client);
    UA_ClientConfig_setDefault(cc);
    cc->customDataTypes = &customDataTypes;
```

 After reading the value I get the ExtensionObject but can't see any relation to my custom datatype. 

```
    retval = UA_Client_readValueAttribute(client, nodeId, &value);
    if(retval == UA_STATUSCODE_GOOD) {
        CustomDataStruct *p = (CustomDataStruct *)value.data;
    }
```

It seems to be a problem with my definition of my custom datatype, so when I read the value my client don't know that this variable is from my custom datatype.
Is that the right way to do? Or is the example only for creating both server and client with open62541. When not, how can I access data from the struct? 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 2d2b98b
 - [x] Other OPC UA SDKs used (client or server): server Siemens PLC , client open62541
 - [x] Operating system: windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

bnadem, 2020-07-17:
i have the same problem , i don'T know customdatatyp as client, how to read datatyep from extentionsobjhect struct??

matkonnerth, 2020-07-18:
Hi @skubidu1324 , can you provide more information about the datatype on server side, maybe an xml nodeset or the bsd file?

This looks strange for me: UA_NODEID_STRING(3, "VT_\"CustomDataStruct\""), /*nodeid from root/Types/VariableTypes/BaseVariableType/*/

I think the nodeId of the datatype should go in here (I think this is the nodeId from the variableType), should be somewhere below root/Types/DataTypes/Structure. Is the binary encoding id really 1, it can be looked up by browsing with for example the UA_Expert from the Datatype and follow the HasEncoding references.

jpfr, 2020-07-18:
A datatype has multiple nodeids.
One used in the information model. Eg. To say that a variable has a certain type.

The other nodeid is for the different encoding formats.
Th extensionobject you received contains the nodeid of the datatype encoding to identify it.
You ought to find it with a debugger. 

So far we „got away“ with using only numeric encoring nodeids for the encodings.
But it seems we have to take the hit in terms of memory and compute time.
And allow all kinds of nodeids to identify binary encodings.

skubidu1324, 2020-07-20:
Thanks @matkonnerth and @jpfr for the quick replies!

I could found the node below `DataTypes`. The reference `HasEncoding` is a node in `(ns=3)` (also string based). This node owns a reference `HasTypeDefinition` with target `DataTypeEncodingType` `(ns=0, i=76)`. Is this the binary encoding id?

@jpfr Do you mean by "So far we „got away“ with using only numeric nodeids for the encodings." that the second parameter of UA_DataType has to be a numric nodeid? So that in my case there is no solution for encoding custom datatype since this nodeid is string based?

jpfr, 2020-07-21:
The fix is in #3795.

skubidu1324, 2020-07-22:
I appreciate your efforts to bring me in the right direction!

Unfortunately I still don't get it working. I will give you some more in Information about the nodeset. I don't know how to export the whole nodeset to xml (70k variables) so I'll try to describe it.

`Objects/Types/DataTypes/BaseDataType/Structure/SimaticStructures`  has a reference `HasSubtype / DT_CustomDataStruct` with nodeid `ns=3;s=DT_"CustomDataStruct"`

This is the `DataType` of my node that I want to read. 
`ns=3;s=DT_CustomDataStruct"` has a reference `HasEncoding / TE_CustomDataStruct` with nodeid `ns=3;s=TE_"CustomDataStruct"`

I think these nodeids are the ones that I have to take into my `CustomDataType` 

```
static const UA_DataType CustomDataStructType = {
	UA_TYPENAME("CustomDataStruct")         
	UA_NODEID_STRING(3, "DT_\"CustomDataStruct\""), 
	UA_NODEID_STRING(3, "TE_\"CustomDataStruct\""), 
	sizeof(CustomDataStruct),					 
	0,                                    
	UA_DATATYPEKIND_STRUCTURE,            
	false,                              
	false,                                 
	16,                                     
	CustomDataStruct_members
};
```
Is that right?

I am left wondering that in the `Fields` attribute of `ns=3;s=DT_CustomDataStruct"` every element of the array is of `DataType` `NodeId` but if I browse in `Objects`  (with UAExpert) and open my struct then I can see the 16 variables with `DataType Boolean`.

matkonnerth, 2020-07-22:
I think that you are right with the typeId and the binaryEncodingId. The attribute "Fields" of the DataType should contain 16 StructureFields, each of them with a member "DataType" with Type Nodeld and value namespace=0, identifier=1(Boolean).

It's interesting that if you browse in Objects, that also the memberfields are expressed as variables. Is it like the ServerStatus Variable (i=2256). There you can read the value direct from the Variable (with DataType ServerStatusDataType), or you can browse it and read for example only the state.

Is UAExpert able to read the value attribute of the variable with the DataTypeDefinition ns=3;s=DT_"CustomDataStruct"?

skubidu1324, 2020-07-22:
> I think that you are right with the typeId and the binaryEncodingId. The attribute "Fields" of the DataType should contain 16 StructureFields, each of them with a member "DataType" with Type Nodeld and value namespace=0, identifier=1(Boolean).

Perfect, this is the same here.

> It's interesting that if you browse in Objects, that also the memberfields are expressed as variables. Is it like the ServerStatus Variable (i=2256). There you can read the value direct from the Variable (with DataType ServerStatusDataType), or you can browse it and read for example only the state.

Yes, it's the same. The node (with my `CustomDataType`) has 16 references `HasComponent` and one `HasTypeDefinition` like `ServerStatus`.

> Is UAExpert able to read the value attribute of the variable with the DataTypeDefinition ns=3;s=DT_"CustomDataStruct"?

Yes,  below `Value` is an other `Value`.  And below that I can see my 16 boolean variables with their values.

matkonnerth, 2020-07-22:
> Yes, below `Value` is an other `Value`. And below that I can see my 16 boolean variables with their values.

Ok, and when you try to read this value with _UA_Client_readValueAttribute_, what is the returned StatusCode?

skubidu1324, 2020-07-22:
I get an `UA_STATUSCODE_GOOD`. But the the UA_Variant which I get has no coherent information

[image]

The typeId and the binaryEncodingId have no content. And casting `data` to my custom stuct gives me just arbitrary values.

```
if (retval == UA_STATUSCODE_GOOD) {
	CustomDataStruct *MyCDS = (CustomDataStruct*)value.data;
}
```

matkonnerth, 2020-07-22:
ok, that seems strange. I've tested it with the PR of @jpfr , the server side seems to work, I've added a struct with two boolean members and gave the type a string id and string encoding id. I can see the correct values in UA_Expert. when I try to read with the client, statuscode returns good but the content is not plausible, no valid typeId. I will dig into this later.

matkonnerth, 2020-07-22:
https://github.com/jpfr/open62541/blob/d366de2433d887897437515a4522a0a7e6596970/src/ua_types_encoding_binary.c#L794
These 2 lines should be removed, then it should also work with string nodeIds.

skubidu1324, 2020-07-23:
Awesome, that's it!!! The returned value has the right `typeId` and  `binaryEncodingId`, so that `data` has my structure inside. Thank you so much for your assistance!  

Now I want to go one step further and read an array of `CustomDataStruct`. Unfortunately the returned value has no `typeId` and  `binaryEncodingId`. But for every element he finds the correct DataType in [here](https://github.com/jpfr/open62541/blob/d366de2433d887897437515a4522a0a7e6596970/src/ua_types_encoding_binary.c#L808).

I think [here](https://github.com/jpfr/open62541/blob/d366de2433d887897437515a4522a0a7e6596970/src/ua_types_encoding_binary.c#L469) is something missing. He goes into [this](https://github.com/jpfr/open62541/blob/d366de2433d887897437515a4522a0a7e6596970/src/ua_types_encoding_binary.c#L1115) function and work here with the wrong type. E.g. he [allocates the wrong size ](https://github.com/jpfr/open62541/blob/d366de2433d887897437515a4522a0a7e6596970/src/ua_types_encoding_binary.c#L495) because  `type->memSize` is not properly filled in that case.

matkonnerth, 2020-07-23:
When reading the array of CustomDataStruct there should be a Variant with array of ExtensionObject returned. You can access then the CustomDataStruct with extensionObj->decoded->data, extensionObj->decoded->type->typeId should contain the right TypeId.

skubidu1324, 2020-07-23:
Yeah, works perfectly! Just in case someone else need it:

```
if (retval == UA_STATUSCODE_GOOD)  {
	UA_ExtensionObject* eo = ((UA_ExtensionObject *)value.data);
	for (int i = 0; i < iArrayLength; i++) 
		CustomDataStruct* MyCDS = (CustomDataStruct*)eo[i].content.decoded.data;
}
```

@matkonnerth  and @jpfr  you're awesome! Thanks for helping me out!

closed 2020-07-23 by skubidu1324