open62541/open62541 #3789: Could not remove mDNS record

ar8384, 2020-07-17:
## Description
Hello,
I am trying to set up a LDS. I tried the code given in open62541/example/discovery/. When a server exits, it seems that the LDS doesn't unregister it correctly.

## Background Information / Reproduction Steps

I launched `server_lds.c`, then `server_register.c`, then `client_find_servers.c`, then I exited the client, I exited `server_register.c`, I launched `client_find_servers.c` again.

Here are the logs of `server_lds.c`, `server_register.c` and `client_find_servers.c` (in that order):

```
demo@demo-Latitude-5500:~/Documents/discovery$ ./lds
[2020-07-17 16:55:14.513 (UTC+0200)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2020-07-17 16:55:14.513 (UTC+0200)] info/server	AccessControl: Anonymous login is enabled
[2020-07-17 16:55:14.513 (UTC+0200)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2020-07-17 16:55:14.513 (UTC+0200)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-07-17 16:55:14.513 (UTC+0200)] warn/network	Multicast DNS: outbound interface 0.0.0.0, it means that the first OS interface is used (you can explicitly set the interface by using 'discovery.mdnsInterfaceIP' config parameter)
[2020-07-17 16:55:14.513 (UTC+0200)] info/network	TCP network layer listening on opc.tcp://demo-Latitude-5500:4840/
[2020-07-17 16:55:14.513 (UTC+0200)] info/server	Multicast DNS: add record for domain: LDS-demo-Latitude-5500._opcua-tcp._tcp.local.
[2020-07-17 16:55:29.380 (UTC+0200)] info/network	Connection 6 | New connection over TCP from 127.0.0.1
[2020-07-17 16:55:29.380 (UTC+0200)] info/channel	Creating a new SecureChannel
[2020-07-17 16:55:29.381 (UTC+0200)] info/channel	Connection 6 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2020-07-17 16:55:29.381 (UTC+0200)] info/server	Multicast DNS: add record for domain: Sample Server-demo-Latitude-5500._opcua-tcp._tcp.local.
[2020-07-17 16:55:36.938 (UTC+0200)] info/network	Connection 7 | New connection over TCP from 127.0.0.1
[2020-07-17 16:55:36.939 (UTC+0200)] info/channel	Creating a new SecureChannel
[2020-07-17 16:55:36.939 (UTC+0200)] info/channel	Connection 7 | SecureChannel 2 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2020-07-17 16:55:36.940 (UTC+0200)] info/channel	Connection 7 | SecureChannel 2 | CloseSecureChannel
[2020-07-17 16:55:36.940 (UTC+0200)] info/network	Connection 7 | Closed
[2020-07-17 16:55:36.940 (UTC+0200)] info/network	Connection 7 | New connection over TCP from 127.0.0.1
[2020-07-17 16:55:36.940 (UTC+0200)] info/channel	Creating a new SecureChannel
[2020-07-17 16:55:36.940 (UTC+0200)] info/channel	Connection 7 | SecureChannel 3 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2020-07-17 16:55:36.941 (UTC+0200)] info/channel	Connection 7 | SecureChannel 3 | CloseSecureChannel
[2020-07-17 16:55:36.941 (UTC+0200)] info/network	Connection 7 | Closed
[2020-07-17 16:55:36.941 (UTC+0200)] info/network	Connection 7 | New connection over TCP from 127.0.0.1
[2020-07-17 16:55:36.941 (UTC+0200)] info/channel	Creating a new SecureChannel
[2020-07-17 16:55:36.942 (UTC+0200)] info/channel	Connection 7 | SecureChannel 4 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2020-07-17 16:55:36.942 (UTC+0200)] info/channel	Connection 7 | SecureChannel 4 | CloseSecureChannel
[2020-07-17 16:55:36.942 (UTC+0200)] info/network	Connection 7 | Closed
[2020-07-17 16:56:44.095 (UTC+0200)] info/server	Multicast DNS: remove record for domain: Sample Server-demo-Latitude-5500._opcua-tcp._tcp.local.
[2020-07-17 16:56:44.095 (UTC+0200)] info/server	Multicast DNS: remove record for domain: Sample Server-demo-Latitude-5500._opcua-tcp._tcp.local.
[2020-07-17 16:56:44.095 (UTC+0200)] warn/server	Could not remove mDNS record for hostname Sample Server.
[2020-07-17 16:56:44.096 (UTC+0200)] info/channel	Connection 6 | SecureChannel 1 | CloseSecureChannel
[2020-07-17 16:56:44.096 (UTC+0200)] info/network	Connection 6 | Closed
[2020-07-17 16:56:56.274 (UTC+0200)] info/network	Connection 6 | New connection over TCP from 127.0.0.1
[2020-07-17 16:56:56.274 (UTC+0200)] info/channel	Creating a new SecureChannel
[2020-07-17 16:56:56.275 (UTC+0200)] info/channel	Connection 6 | SecureChannel 5 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2020-07-17 16:56:56.275 (UTC+0200)] info/channel	Connection 6 | SecureChannel 5 | CloseSecureChannel
[2020-07-17 16:56:56.276 (UTC+0200)] info/network	Connection 6 | Closed
[2020-07-17 16:56:56.276 (UTC+0200)] info/network	Connection 6 | New connection over TCP from 127.0.0.1
[2020-07-17 16:56:56.276 (UTC+0200)] info/channel	Creating a new SecureChannel
[2020-07-17 16:56:56.276 (UTC+0200)] info/channel	Connection 6 | SecureChannel 6 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2020-07-17 16:56:56.277 (UTC+0200)] info/channel	Connection 6 | SecureChannel 6 | CloseSecureChannel
[2020-07-17 16:56:56.277 (UTC+0200)] info/network	Connection 6 | Closed
[2020-07-17 16:56:56.277 (UTC+0200)] info/network	Connection 6 | New connection over TCP from 127.0.0.1
[2020-07-17 16:56:56.278 (UTC+0200)] info/channel	Creating a new SecureChannel
[2020-07-17 16:56:56.278 (UTC+0200)] info/channel	Connection 6 | SecureChannel 7 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2020-07-17 16:56:56.279 (UTC+0200)] info/channel	Connection 6 | SecureChannel 7 | CloseSecureChannel
[2020-07-17 16:56:56.279 (UTC+0200)] info/network	Connection 6 | Closed
^C[2020-07-17 16:57:13.116 (UTC+0200)] info/network	Shutting down the TCP network layer
[2020-07-17 16:57:13.116 (UTC+0200)] info/server	Multicast DNS: remove record for domain: LDS-demo-Latitude-5500._opcua-tcp._tcp.local.
```

```
demo@demo-Latitude-5500:~/Documents/discovery$ ./server 
[2020-07-17 16:55:28.878 (UTC+0200)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2020-07-17 16:55:28.878 (UTC+0200)] info/server	AccessControl: Anonymous login is enabled
[2020-07-17 16:55:28.878 (UTC+0200)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2020-07-17 16:55:28.878 (UTC+0200)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-07-17 16:55:28.878 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No TypeDefinition for ns=1;s=the.answer; Use the default TypeDefinition for the Variable/Object
[2020-07-17 16:55:28.878 (UTC+0200)] info/userland	Node read the.answer
[... 156 lines cut ...]
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_PARSING:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=REDUCED
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (1.1.1):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 20.04 LTS
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ar8384, 2020-07-24:
I may have found the problem in server_register.c:

Instead of (line 102)
```C
UA_StatusCode retval = UA_Server_addPeriodicServerRegisterCallback(server, clientRegister, DISCOVERY_SERVER_ENDPOINT, 10 * 60 * 1000, 500, NULL);
```
shouldn't it be:
```C
UA_UInt64 callbackId;
UA_StatusCode retval = UA_Server_addPeriodicServerRegisterCallback(server, clientRegister, DISCOVERY_SERVER_ENDPOINT, 10 * 60 * 1000, 500, &callbackId);
```
and then add after `UA_Server_unregister_discovery` (line 133):
```
UA_Server_removeCallback(server, callbackId);
```
?

jpfr, 2020-07-26:
That sounds like the solution.
Care to send a PR?

ar8384, 2020-07-27:
Done

jpfr, 2020-07-30:
Nice. Please reopen this one if required.

closed 2020-07-30 by jpfr