open62541/open62541 #3801: [PubSub] UA_Server_addWriterGroup() does not register callback to server

chuongvhn, 2020-07-26:
## Description
with current latest release v1.1.2 , in implementation of UA_Server_addWriterGroup() the call to  UA_WriterGroup_addPublishCallback() was removed so server does not publish when PubSub connection and WriterGroup are added.
the impl in v1.0.0
https://github.com/open62541/open62541/blob/70b87b47fe8300ddbaafbdec34319a71a8640199/src/pubsub/ua_pubsub.c#L162
the impl in v1.1.2
https://github.com/open62541/open62541/blob/8ad84962e67987af91d7f746425d7b26be47a0cb/src/pubsub/ua_pubsub_writer.c#L202

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-07-28:
You have to set the Writer group operational.
This is no longer automatic so you can finish to set the configuration before publishing.

chuongvhn, 2020-08-04:
Yes, we need to call UA_Server_setWriterGroupOperational( server, writer_group_id );
Thank you !

closed 2020-08-04 by chuongvhn