open62541/open62541 #3803: [PubSub] UA_Server_processNetworkMessage() does not process correctly DataSetMessage which has multiple DataSetWriter

chuongvhn, 2020-07-27:
## Description
Publisher publishes multiple DataSet (each one corresponding with a DataSetWriter) in one WriterGroup, Subscriber process DataSetMessage incorrectly. Subscriber always take the first DataSetReader in Server then processes this DataSetReader with all DataSetWriters in received NetworkMessage

Publisher Process Network Message from here: UA_Server_processNetworkMessage()
https://github.com/open62541/open62541/blob/8ad84962e67987af91d7f746425d7b26be47a0cb/src/pubsub/ua_pubsub_reader.c#L679

It will process per each DataSetMessage encapsulated in NetWorkMessage() by UA_Server_DataSetReader_process()
https://github.com/open62541/open62541/blob/8ad84962e67987af91d7f746425d7b26be47a0cb/src/pubsub/ua_pubsub_reader.c#L181
UA_Server_DataSetReader_process() will match DataSetReader which is configured in subscriber's server by getReaderFromIdentifier()
https://github.com/open62541/open62541/blob/8ad84962e67987af91d7f746425d7b26be47a0cb/src/pubsub/ua_pubsub_reader.c#L160
getReaderFromIdentifier() looks though all ReaderGroup to check for ReaderDataSet against DatSetReader configured in Subscriber by calling checkReaderIdentifier()
https://github.com/open62541/open62541/blob/8ad84962e67987af91d7f746425d7b26be47a0cb/src/pubsub/ua_pubsub_reader.c#L160
checkReaderIdentifier() will return the first match DataSetReader in Subscriber and DataSetWriter in received NetworkMessage.
and use this Reader to process for all DataSetMessage.
E.g: received NetworkMessage {DataSetWriterA, DatsSetWriter B}  
Subscriber configuration has {DataSetReaderA, DataSetReaderB}
checkReaderIdentifier() will return DataSetReaderA.
thend DataSetReaderA will be used to process DataSetMessageA , DataSetMessageB (BadTyeMismatch)

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

chuongvhn, 2020-08-05:
#3823

NoelGraf, 2021-04-30:
#3887