open62541/open62541 #3805: How UA_Client_getEndpoints() get encryption endpoints?

jixian79, 2020-07-27:
## Description
I have a opcua server with `basic256sha256/None` Security Policy, `username` user Identity token,  `Sign&Encrypt` Message Security， the server has registed to LDS 。And **the server only have a  basic256sha256 endpoint.**

UaExpert can find my opcua server with basic256sha256 endpoint。
But UA_Client_getEndpoints() can't find the endpoint.

the opcua server set the `securityPolicyNoneDiscoveryOnly`  true

How can i get the basic256sha256 endpoint with `UA_Client_getEndpoints()?`

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

-  [x] open62541 Version (release number or git tag): v 1.1.1
 - [x] Other OPC UA SDKs used (client or server):  open62541's server and client, UaExpert
 - [x] Operating system:Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ZekeLandon, 2020-08-13:
@jixian79, could you post your logs here related to UA_Client_getEndpoints() and the server for better understanding?
When UaExpert can find the endpoints, internal client should also be able to do so.

NoelGraf, 2021-04-30:
@jixian79 Is your issue still current?

NoelGraf, 2021-05-20:
I would then close the issue. Please reopen if required!

closed 2021-05-20 by NoelGraf