open62541/open62541 #3807: Question about UA_RegisterNodesRequest

luibass92, 2020-07-27:
## Description

Hello everyone,

I have a question regarding `UA_RegisterNodesRequest `and `UA_RegisterNodesResponse`.
I am trying to communicate with a Siemens S7-1500 PLC and I have seen that it supports Registered Nodes Requests in order to speed up frequently readings of the registered nodes.

I found in open62541 documentation that the types I am interested in, exist. But still I cannot figure out how to correctly perform a RegisterNodes request from a client.

I found the API:

```
/**
 * RegisterNodes Service
 * ^^^^^^^^^^^^^^^^^^^^^
 * Used by Clients to register the Nodes that they know they will access
 * repeatedly (e.g. Write, Call). It allows Servers to set up anything needed so
 * that the access operations will be more efficient. */
void Service_RegisterNodes(UA_Server *server, UA_Session *session,
                           const UA_RegisterNodesRequest *request,
                           UA_RegisterNodesResponse *response);
```

But I did not get how to use it. The description says "used by clients" but I see that the imput parameter is the server (?).

Someone can give me some tips about it?

Thanks.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.1.2
 - [ ] Operating system: Windows 10
 - [ ] Self-contained code example attached

opcua-tsn-team-kalycito, 2020-07-28:
Hi @luibass92,

The API `Service_RegisterNodes`, is internal to server. 
Once a client sends a `RegisterNodesRequest`,
1.	The `getServicePointers` function, decodes the request
a.	If the requested service by the client is `RegisterNodesRequest`,
b.	then it maps it to the `Service_RegisterNodes` function and fills in the required input arguments for this API

For example,
1.	On the client side, we are registering the DateTime variable node (ns=0;i=2258)
>     UA_RegisterNodesRequest req;
>     UA_RegisterNodesRequest_init(&req);
>     req.nodesToRegister = UA_NodeId_new();
>     req.nodesToRegister[0] = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_ CURRENTTIME);
>     req.nodesToRegisterSize = 1;
>     UA_RegisterNodesResponse res = UA_Client_Service_registerNodes(client, req);
2.	The request to register that node is send to server. Refer to the screenshots below
[image]
3.	The server decodes the same and makes use of the `getServicePointers `function to map to the requested service
4.	It fills in the required arguments and then executes the `Service_RegisterNodes`

Does this help with the tips that you are looking for?

luibass92, 2020-07-28:
> UA_Client_Service_registerNodes

Thanks @opcua-tsn-team-kalycito now it is clear!
I did not find the API `UA_Client_Service_registerNodes` 

Thanks again, bye!

opcua-tsn-team-kalycito, 2020-07-28:
You're welcome, do you think this issue can be closed now?

luibass92, 2020-07-28:
> You're welcome, do you think this issue can be closed now?

Sure.

closed 2020-07-28 by luibass92