open62541/open62541 #3808: volatile compound assignment deprecated in c++20

mlgiraud, 2020-07-28:
## Description
c++20 deprecates some uses of volatile, which breaks compilation when compiling the library with a c++20 compiler (e.g. clang11)
This is used in [include/open62541/architecture_definitions.h](https://github.com/open62541/open62541/blob/629d9ddd1dd2a9cf18eb1afd4134648da48cf0ee/include/open62541/architecture_definitions.h#L424) for example.
The error:
```
warning: compound assignment to object of volatile-qualified type 'volatile uint32_t' (aka 'volatile unsigned int') is deprecated [-Wdeprecated-volatile]
    *addr += increase;
```

Solution: Disable the warning via flag, or replace `a+=b` with `a=a+b`

## Background Information / Reproduction Steps
compile with clang11 as c++

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2020-07-28:
Duplicate of #3712

mlgiraud, 2020-07-28:
Closing in favor of continuing in the original issue

closed 2020-07-28 by mlgiraud