open62541/open62541 #3810: Error opening server socket

Sarutobie, 2020-07-28:
## Description
I could succesfully build and install the library.
I then compiled the example tutorial_pubsub_publish.
Everything works fine and i also tested on another vm.

Now i wanted to test the server on another linux system, which is customized with kernel 4.9.20.

If i start the server there i get the error **Join multicast group failed**
After using this line: route add -net 224.0.0.0 netmask 240.0.0.0 dev eth0
I get the **Error opening the server socket**

Can someone tell me what i can look into, to fix the problem?

## Background Information / Reproduction Steps

Used CMake options:
```
UA_ENABLE_HARDENING: ON
UA_ENABLE_METHODCALLS: ON
UA_ENABLE_NODEMANAGEMENT: ON
UA_ENABLE_NODESET_COMPILER_DES: ON
UA_ENABLE_PARSING: ON
UA_ENABLE_PUBSUB: ON
UA_ENABLE_PUBSUB_ETH_UADP: ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL: ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS: ON
UA_ENABLE_STATUSCODE_DESCRIPTIONS: ON
UA_ENABLE_SUBSCRIPTIONS: ON
UA_ENABLE_TYPEDESCRIPTIONS: ON
UA_NAMESPACE_ZERO: FULL
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: customized linux build kernel 4.9.20
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Sarutobie, 2020-07-29:
I did some debugging on my own and the actual error is **Address family not supported by protocol**
In this case the addrinfo looks like this:
family: 10
type : 1
protocol: 6
flags: 1

Now i have to look why this error is being caused

jpfr, 2020-07-29:
Looks like you should disable IPV6.
Your architecture port probably sets the `UA_IPV6` macro.

Sarutobie, 2020-07-30:
I've disabled UA_IPV6 in architecture.
The error now changed to unknown error -1

I looked up address-family and 10 seems to be CCITT protocols.
So it seems the linux i am using have some problems with a CCITT protocol.
X.509 for example is also in this group.

gc87, 2020-08-07:
I have the same problem，CUP：TI-am3354，armv8，Linux：3.6. I've disabled UA_IPV6. Problem still exists.

jpfr, 2020-08-13:
Hmm. We could just use AF_INET if ipv6 is disabled. Care to provide a PR?

GrantEdwards, 2020-09-15:
My fix for this problem is to add the following code at the top of addSeverSocket() in network_tcp.c:

static void
addServerSocket(ServerNetworkLayerTCP *layer, struct addrinfo *ai) {

#if !UA_IPV6
  if(ai->ai_family == AF_INET6)
    return;
#endif 
 
    /* Create the server socket */

GrantEdwards, 2020-09-15:
The underlying error is assuming that all address families returned by a call to getaddrinfo() must be supported by the network stack.  That's not true.  It's not unusual to have a networking libarary and DNS system that will return both IPv4 and IPv6 addresses for a hostname even though the kernel and network stack do not have IPv6 support.

teccheck, 2025-05-20:
Because of the event loop, network code is quite different now, so I'll close this

closed 2025-05-20 by teccheck