open62541/open62541 #3811: Write to specific array position

JulZimmermann, 2020-07-30:
## Description

I try to write a single value in an array. Because there is no documentation or example given to do so, I do it as described In #803.

I get a zero error code in my UA_WriteResponse, but the value on the server doesn't change.

## Background Information / Reproduction Steps

Used Code:

```
bool boolValue = true;
UA_Variant value;
UA_Variant_init(&value);
UA_Variant_setArray(&value, &boolValue, 1, &UA_TYPES[UA_TYPES_BOOLEAN]);

UA_WriteValue wValue;
UA_WriteValue_init(&wValue);
wValue.nodeId = nodeId;
wValue.attributeId = UA_ATTRIBUTEID_VALUE;
wValue.value.value = value;
wValue.value.hasValue = true;
wValue.indexRange = UA_STRING("0");

UA_WriteRequest wReq;
UA_WriteRequest_init(&wReq);
wReq.nodesToWrite = &wValue;
wReq.nodesToWriteSize = 1;

UA_WriteResponse wResp = UA_Client_Service_write(client, wReq);
UA_WriteResponse_deleteMembers(&wResp);
```

Output with UA_LOGLEVEL set to 0:

```
[2020-07-30 09:42:53.727 (UTC+0200)] debug/channel	Connection 524 | SecureChannel 4 | Sending request with RequestId 6 of type WriteRequest
[2020-07-30 09:42:53.732 (UTC+0200)] debug/client	Decode a message of type WriteResponse
```

Used CMake options:

```bash
cmake -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DOPEN62541_VERSION=v1.1.2 ..
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.1.2
 - [X] Other OPC UA SDKs used (client or server): Softing Demo Server
 - [X] Operating system: Windows and Linux (WSL)
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-07-30:
So you use a Softing Demo Server.
Does the value change work with a different client (e.g. UAExpert)?

JulZimmermann, 2020-07-31:
As far as I see, UAExpert doesn't support writing to a specific array position. With this Client, I can only write the whole array.

jpfr, 2020-07-31:
It could be unsupported server-side.
Please test with any other closed or open source client to verify it works for them.

JulZimmermann, 2020-09-22:
Sorry for the late response.

I tried with an open62541 based server and it seems to work correctly. Will close the item

closed 2020-09-22 by JulZimmermann