open62541/open62541 #3814: Not able to connect with server using UA_Client_connect

Kaushil17, 2020-07-30:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_ENABLE_ENCRYPTION=ON ..
```

## Checklist

Please provide the following information:

 - open62541 Version (release number or git tag): 691a7af
 - Other OPC UA SDKs used (client or server): client
 - Operating system: Ubuntu 20.04.1 LTS 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - Self-contained code example attached
 - [ ] Critical issue

This is my code.
```UA_Client *client = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client));
    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
   if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return EXIT_FAILURE;
    }
/* Read the server-time */
    UA_Variant value;
    UA_Variant_init(&value);
    UA_Client_readValueAttribute(client,
              UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME),
              &value);
    if(UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_DATETIME])) {
        UA_DateTimeStruct dts = UA_DateTime_toStruct(*(UA_DateTime *)value.data);
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                    "The server date is: %02u-%02u-%04u %02u:%02u:%02u.%03u",
                    dts.day, dts.month, dts.year, dts.hour, dts.min, dts.sec, dts.milliSec);
    }
    UA_Variant_clear(&value);

    UA_Client_disconnect(client);
    UA_Client_delete(client);```

I am getting these errors:
```[2020-07-30 22:44:30.772 (UTC+0550)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-07-30 22:44:30.779 (UTC+0550)] info/channel	Connection 12 | SecureChannel 22 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2020-07-30 22:44:30.779 (UTC+0550)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2020-07-30 22:44:30.781 (UTC+0550)] info/client	Rejecting UserTokenPolicy 0 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
[2020-07-30 22:44:30.781 (UTC+0550)] info/client	Rejecting UserTokenPolicy 1 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
[2020-07-30 22:44:30.781 (UTC+0550)] error/client	No suitable UserTokenPolicy found for the possible endpoints
[2020-07-30 22:44:30.781 (UTC+0550)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError
```

Any change i need to do in code for this?

jpfr, 2020-07-30:
You have to enable encryption in the config.
E.g. set your certificate and private key for encryption.

See the examples.

closed 2020-07-30 by jpfr

Kaushil17, 2020-07-31:
> ## Description
> ## Background Information / Reproduction Steps
> Used CMake options:
> 
> ```shell
> cmake -DUA_ENABLE_ENCRYPTION=ON ..
> ```
> 
> ## Checklist
> Please provide the following information:
> 
> * open62541 Version (release number or git tag): [691a7af](https://github.com/open62541/open62541/commit/691a7af9ac7f2a0df5e2a6e449dffea46a8e3714)
> * Other OPC UA SDKs used (client or server): client
> * Operating system: Ubuntu 20.04.1 LTS
> * [ ]  Logs (with `UA_LOGLEVEL` set as low as necessary) attached
> * [ ]  Wireshark network dump attached
> * Self-contained code example attached
> * [ ]  Critical issue
> 
> This is my code.
> 
> ```
>     UA_ClientConfig_setDefault(UA_Client_getConfig(client));
>     UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
>    if(retval != UA_STATUSCODE_GOOD) {
>         UA_Client_delete(client);
>         return EXIT_FAILURE;
>     }
> /* Read the server-time */
>     UA_Variant value;
>     UA_Variant_init(&value);
>     UA_Client_readValueAttribute(client,
>               UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME),
>               &value);
>     if(UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_DATETIME])) {
>         UA_DateTimeStruct dts = UA_DateTime_toStruct(*(UA_DateTime *)value.data);
>         UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
>                     "The server date is: %02u-%02u-%04u %02u:%02u:%02u.%03u",
>                     dts.day, dts.month, dts.year, dts.hour, dts.min, dts.sec, dts.milliSec);
>     }
>     UA_Variant_clear(&value);
> 
>     UA_Client_disconnect(client);
>     UA_Client_delete(client);
> ```
> 
> I am getting these errors:
> 
> ```
> [2020-07-30 22:44:30.772 (UTC+0550)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
> [2020-07-30 22:44:30.779 (UTC+0550)] info/channel	Connection 12 | SecureChannel 22 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
> [2020-07-30 22:44:30.779 (UTC+0550)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
> [2020-07-30 22:44:30.781 (UTC+0550)] info/client	Rejecting UserTokenPolicy 0 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
> [2020-07-30 22:44:30.781 (UTC+0550)] info/client	Rejecting UserTokenPolicy 1 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
> [2020-07-30 22:44:30.781 (UTC+0550)] error/client	No suitable UserTokenPolicy found for the possible endpoints
> [2020-07-30 22:44:30.781 (UTC+0550)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError
> ```
> 
> Any change i need to do in code for this?

After all configuration added i got these errors->
```
[2020-07-31 04:39:54.808 (UTC+0550)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-07-31 04:39:54.808 (UTC+0550)] info/securitypolicy	The Basic128Rsa15 security policy with openssl is added.
[2020-07-31 04:39:54.810 (UTC+0550)] info/securitypolicy	The basic256 security policy with openssl is added.
[2020-07-31 04:39:54.810 (UTC+0550)] info/securitypolicy	The basic256sha256 security policy with openssl is added.
[2020-07-31 04:39:54.810 (UTC+0550)] info/securitypolicy	The Aes128Sha256RsaOaep security policy with openssl is added.
[2020-07-31 04:39:54.810 (UTC+0550)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None                                               
[2020-07-31 04:39:54.823 (UTC+0550)] info/channel	Connection 3 | SecureChannel 7 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2020-07-31 04:39:54.823 (UTC+0550)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2020-07-31 04:39:54.834 (UTC+0550)] info/client	Rejecting endpoint 0: security mode doesn't match                                                   
[2020-07-31 04:39:54.834 (UTC+0550)] info/client	Selected Endpoint opc.tcp://kaushil:53530/OPCUA/SimulationServer with SecurityMode Sign and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2020-07-31 04:39:54.834 (UTC+0550)] info/client	Selected UserTokenPolicy certificate_basic256 with UserTokenType Certificate and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2020-07-31 04:39:54.834 (UTC+0550)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good                                                 
[2020-07-31 04:39:54.835 (UTC+0550)] info/securitypolicy	The basic256 security policy channel with openssl is created.
[2020-07-31 04:39:54.922 (UTC+0550)] info/channel	Connection 3 | SecureChannel 8 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256 and a revised lifetime of 600.00s
[2020-07-31 04:39:54.922 (UTC+0550)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2020-07-31 04:39:54.954 (UTC+0550)] info/client	Received a ServiceFault response
[2020-07-31 04:39:54.954 (UTC+0550)] info/client	The ServiceResult has the StatusCode BadInternalError
[2020-07-31 04:39:54.954 (UTC+0550)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: BadInternalError
[2020-07-31 04:39:54.954 (UTC+0550)] info/securitypolicy	The basic256 security policy channel with openssl is deleted.
[2020-07-31 04:39:54.954 (UTC+0550)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError
```

waltronix, 2020-08-04:
I have similar problems when using the example client with OpenSSL. Therefore I created certificates with the provided python script.
The example server works fine. I was connecting to it with the prosys client

So far, I haven't tried mbed TLS. That would have been my next step.

waltronix, 2020-08-04:
I rebuild everything using mbed TLS. 
- created two pairs of certificates, one for the server, one for client.
- run server_ctt.exe with one pair of certificate & key
  - connecting to the server using prosys client works
  - disconnected prosys client
- client client_encryption.exe using second pair of certificate & key fails:

```
[2020-08-04 20:44:52.360 (UTC+0200)] info/client        Selected Endpoint opc.tcp://host.docker.internal:53530/OPCUA/SimulationServer with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2020-08-04 20:44:52.362 (UTC+0200)] info/client        Selected UserTokenPolicy anonymous with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2020-08-04 20:44:52.363 (UTC+0200)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2020-08-04 20:44:52.364 (UTC+0200)] trace/client       Client connect iterate
[2020-08-04 20:44:52.366 (UTC+0200)] trace/client       Client connect iterate
[2020-08-04 20:44:52.367 (UTC+0200)] trace/client       Client connect iterate
[2020-08-04 20:44:52.367 (UTC+0200)] debug/network      Sent HEL message
[2020-08-04 20:44:52.368 (UTC+0200)] debug/client       Client Status: ChannelState: HELSent, SessionState: Closed, ConnectStatus: Good
[2020-08-04 20:44:52.369 (UTC+0200)] trace/client       Client connect iterate
[2020-08-04 20:44:52.372 (UTC+0200)] debug/network      Received ACK message
[2020-08-04 20:44:52.373 (UTC+0200)] debug/client       Client Status: ChannelState: AckReceived, SessionState: Closed, ConnectStatus: Good
[2020-08-04 20:44:52.373 (UTC+0200)] trace/client       Client connect iterate
[2020-08-04 20:44:52.374 (UTC+0200)] debug/channel      Connection 376 | SecureChannel 0 | Generating new local nonce
[2020-08-04 20:44:52.376 (UTC+0200)] debug/channel      Connection 376 | SecureChannel 0 | Requesting to open a SecureChannel
[2020-08-04 20:44:52.377 (UTC+0200)] debug/channel      Requesting to open a SecureChannel
[2020-08-04 20:44:52.409 (UTC+0200)] debug/channel      OPN message sent
[2020-08-04 20:44:52.410 (UTC+0200)] debug/client       Client Status: ChannelState: OPNSent, SessionState: Closed, ConnectStatus: Good
[2020-08-04 20:44:52.410 (UTC+0200)] trace/client       Client connect iterate
[2020-08-04 20:44:52.416 (UTC+0200)] error/channel      Connection 376 | SecureChannel 0 | Received an ERR response with StatusCode BadSecurityChecksFailed and the following reason: Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security.")
[2020-08-04 20:44:52.418 (UTC+0200)] warn/channel       Connection 376 | SecureChannel 0 | Receiving the response failed with StatusCode BadConnectionClosed
[2020-08-04 20:44:52.420 (UTC+0200)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadConnectionClosed
[2020-08-04 20:44:52.422 (UTC+0200)] debug/securitypolicy       Deleted members of EndpointContext for sp_basic128rsa15
[2020-08-04 20:44:52.423 (UTC+0200)] debug/securitypolicy       Deleted members of EndpointContext for sp_basic256
[2020-08-04 20:44:52.423 (UTC+0200)] debug/securitypolicy       Deleted members of EndpointContext for sp_basic256sha256
```

The result here is the same as when using OpenSSL.