open62541/open62541 #3817: Multi-threaded async client

xallaume, 2020-07-31:
Hello,

My app shall send service requests to an OPC server asynchronously. Additionally, the call to UA_Client_sendAsyncRequest is performed on dynamically allocated threads, not the one where the UA_Client is instantiated.

The call to UA_Client_sendAsyncRequest and subsequent UA_Client_run_iterate(client, 0) is of course protected by a mutex. I can see that UA_Client_sendAsyncRequest call is taken into consideration by the client, but the UA_Client_run_iterate call doesn't trigger the response callback. In order to get this, I need to provide a value superior to 0 in UA_Client_run_iterate. But from my understanding, this delays the firig of the callback by the amount of time provided in UA_Client_run_iterate, which is kind of contradicting with the async purpose (efficency).

Is there something I didn't understand properly that implies this situation? Considering my experience with open62541, there's probably a lot! Anyhow, congratulations for the wonderful work!

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): : 1.1.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: KUbuntu 19.10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlgiraud, 2020-08-06:
I'm not sure i understand you correctly. But as far as i understood, you mean that the run_iterate call blocks for the full duration supplied in the timeout value, regardless of how many async responses were received during this time?
If this is the case, then that behavior is not intended afaik and should be fixed once the networkmanager lands.
@jpfr What do you think about this?

jpfr, 2020-08-13:
Hmm. A timeout of zero *should* work as intended: process the received messages but don’t sleep to receive more.

If not this is likely a bug in the network layer.

teccheck, 2025-05-21:
The whole server API is now thread safe and the relevant mechanism is overhauled. Therefore, I'll close this. Feel free to reopen if you need