open62541/open62541 #3821: How to get OPC UA variable nodes of the PLC, to address them in an OPC UA client-open62541

mohamedalielghoula, 2020-08-04:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

embbo, 2020-08-13:
@mohamedalielghoula what do yu mean ? what i understand you are asking about Plc adress ..in VariableNode?

mohamedalielghoula, 2020-08-13:
Hello @embbo,
yes exactly

jpfr, 2020-08-13:
Connect with a graphical client like UA Expert.
Or use the browse service to find nodes.

mohamedalielghoula, 2020-08-17:
Hello @jpfr , I'm a beginner in open62541 
in my project, I use two stm32s 
the first includes an opc client to communicate with an opc server in PLC and the second includes an opc server to communicate with a desktop application opc ua, the communication between the two cards is serial, 
I don't need all the node IDs in the PLC, so I access the node IDs in the first board by names that are not with the navigation service, and I use this function for each node ID :
 UA_Client_readValueAttribute(client, UA_NODEID_STRING(3, "xxx"), &value) ;
My question is that there are some nodes under a repositories named like this xx.yy.zz 
how do I get it with this function?

embbo, 2020-08-17:
@jpfr can I write somewhere Userdata ?? for example I need tou send adress of each Node with that Node (0x1234456), where can I write it ..

embbo, 2020-08-17:
@mohamedalielghoula ..what i am doing , I use two node . and I put adress in value of the other..

mohamedalielghoula, 2020-08-19:
@embbo Sorry but I'm not understood, can you please tell me more. 
In my example, my node ID is "FB_DB". "Tag1" and I want to read the value of Tag1, and there are other nodes under FB_DB, 
how can read it with this function UA_Client_readValueAttribute() 
or there is another function that allows me to get access under "FB_DB".

mohamedalielghoula, 2020-08-19:
> Connect with a graphical client like UA Expert.
> Or use the browse service to find nodes.

i have nodeid string separated by point as u see in my example "FB_DB". "Tag1", it didnt work with this function 
UA_Client_readValueAttribute(client, UA_NODEID_STRING(3, "xxx"), &value)

EugenM-sie, 2020-09-18:
@mohamedalielghoula hi, wich desktop application opc ua do you mean?

mohamedalielghoula, 2020-09-25:
@EugenM-sie i developped my own opc ua client application 
The problem is solved
 To read this node i have to named in this function like this UA_NODEID_STRING(3, "\"FB_DB_1\".\"Tag1\"")

closed 2020-09-25 by mohamedalielghoula