open62541/open62541 #3826: ValueRank of -2 (Any) not compatible with a ValueRank of -1 (Scalar) on base type

mgroenhoff, 2020-08-05:
## Description

I am having an issue loading the PA-DIM node set. It defines `DiscreteSignalVariableType` with a ValueRank of -2 but its base type `MultiStateDictionaryEntryDiscreteBaseType` has a ValueRank of -1. Apparently it is not a valid combination as implemented in [compatibleValueRanks()](https://github.com/open62541/open62541/blob/v1.1.2/src/server/ua_services_attribute.c#L770).

Is this an implementation or a spec issue?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Win10
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: see below
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

## Log

```
info/session ... | AddNodes: The value rank of ns=4;i=1142 is incompatible with the value rank of the VariableType
info/session ... | AddNodes: Type-checking the variable node ns=4;i=1142 failed with error code BadTypeMismatch
```

ccvca, 2020-08-06:
Seems like a issue in the specification.

When the BaseType has set UA_VALUERANK_SCALAR (-1), the more specific SubType cannot say UA_VALUERANK_ANY (-2), as the specific type can no longer be used in all places where the BaseType can be used.

But after a short look at the specification, I'm unable to find the issue:
https://github.com/OPCF-Members/UA-NodeSet/blob/e97ac18250124a5229472f80f8df178b99d3767f/PADIM/Opc.Ua.PADIM.NodeSet2.xml#L702-L708
```xml
  <UAVariableType NodeId="ns=3;i=1143" BrowseName="3:DiscreteSignalVariableType" ValueRank="-2">
    <DisplayName>DiscreteSignalVariableType</DisplayName>
    <Documentation>https://reference.opcfoundation.org/v104/PADIM/v100/docs/8.16.2</Documentation>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=2372</Reference>
    </References>
  </UAVariableType>
```

```xml
<UAVariableType NodeId="i=2372" BrowseName="DiscreteItemType" IsAbstract="true" ValueRank="-2">
```

mgroenhoff, 2020-08-06:
Oh i see, sorry. I was referring to the wrong type. I meant to say [MultiStateDiscreteSignalVariableType](https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/PADIM/Opc.Ua.PADIM.NodeSet2.xml#L694) instead of DiscreteSignalVariableType being a subtype of MultiStateDictionaryEntryDiscreteBaseType.

```xml
  <!-- https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/PADIM/Opc.Ua.PADIM.NodeSet2.xml#L694 -->
  <UAVariableType NodeId="ns=3;i=1142" BrowseName="3:MultiStateDiscreteSignalVariableType" DataType="UInt32" ValueRank="-2">
    <DisplayName>MultiStateDiscreteSignalVariableType</DisplayName>
    <Documentation>https://reference.opcfoundation.org/v104/PADIM/v100/docs/8.15</Documentation>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=19077</Reference>
      <Reference ReferenceType="HasDictionaryEntry">ns=1;s=0112/2///61987#ABN636#001</Reference>
    </References>
  </UAVariableType>

  <!-- https://raw.githubusercontent.com/OPCFoundation/UA-Nodeset/v1.04/Schema/Opc.Ua.NodeSet2.xml -->
  <UAVariableType NodeId="i=19077" BrowseName="MultiStateDictionaryEntryDiscreteBaseType" DataType="Number">
    <DisplayName>MultiStateDictionaryEntryDiscreteBaseType</DisplayName>
    <Documentation>https://reference.opcfoundation.org/v104/Core/docs/Part19/7.1</Documentation>
    <References>
      <Reference ReferenceType="HasProperty">i=19082</Reference>
      <Reference ReferenceType="HasProperty">i=19083</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=11238</Reference>
    </References>
  </UAVariableType>

  <!-- https://raw.githubusercontent.com/OPCFoundation/UA-Nodeset/v1.04/Schema/Opc.Ua.NodeSet2.xml -->
  <UAVariableType NodeId="i=11238" BrowseName="MultiStateValueDiscreteType" DataType="Number" ValueRank="-2">
    <DisplayName>MultiStateValueDiscreteType</DisplayName>
    <Documentation>https://reference.opcfoundation.org/v104/Core/docs/Part8/5.3.3/#5.3.3.4</Documentation>
    <References>
      <Reference ReferenceType="HasProperty">i=11241</Reference>
      <Reference ReferenceType="HasProperty">i=11461</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=2372</Reference>
    </References>
  </UAVariableType>
```

However the generated code does not seem to correctly take the ValueRank of the base type. Note that it says that the value is inherited but the value should be -2?

```c
// open62541/src_generated/open62541/namespace0_generated.c
/* MultiStateValueDiscreteType - ns=0;i=11238 */

static UA_StatusCode function_namespace0_generated_1009_begin(UA_Server *server, UA_UInt16* ns) {
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_VariableTypeAttributes attr = UA_VariableTypeAttributes_default;
attr.valueRank = -2;
attr.dataType = UA_NODEID_NUMERIC(ns[0], 26);
attr.displayName = UA_LOCALIZEDTEXT("", "MultiStateValueDiscreteType");
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_VARIABLETYPE,
UA_NODEID_NUMERIC(ns[0], 11238),
UA_NODEID_NUMERIC(ns[0], 2372),
UA_NODEID_NUMERIC(ns[0], 45),
UA_QUALIFIEDNAME(ns[0], "MultiStateValueDiscreteType"),
UA_NODEID_NUMERIC(ns[0], 0),
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_VARIABLETYPEATTRIBUTES],NULL, NULL);
return retVal;
}

/* MultiStateDictionaryEntryDiscreteBaseType - ns=0;i=19077 */

static UA_StatusCode function_namespace0_generated_1012_begin(UA_Server *server, UA_UInt16* ns) {
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_VariableTypeAttributes attr = UA_VariableTypeAttributes_default;
/* Value rank inherited */
attr.valueRank = -1;
attr.dataType = UA_NODEID_NUMERIC(ns[0], 26);
attr.displayName = UA_LOCALIZEDTEXT("", "MultiStateDictionaryEntryDiscreteBaseType");
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_VARIABLETYPE,
UA_NODEID_NUMERIC(ns[0], 19077),
UA_NODEID_NUMERIC(ns[0], 11238),
UA_NODEID_NUMERIC(ns[0], 45),
UA_QUALIFIEDNAME(ns[0], "MultiStateDictionaryEntryDiscreteBaseType"),
UA_NODEID_NUMERIC(ns[0], 0),
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_VARIABLETYPEATTRIBUTES],NULL, NULL);
return retVal;
}

```

Also I can't access the repository you are referring to. Is it the same as https://github.com/OPCFoundation/UA-Nodeset ?

ccvca, 2020-08-06:
> Also I can't access the repository you are referring to. Is it the same as https://github.com/OPCFoundation/UA-Nodeset ?

Sorry, I used the Members-Only repository, yes it's the same for there types. (You may request access here https://github.com/OPCF-Members/Help )

After some investigation, I do believe, that PADIM has a (critical) error and open62541 does everything right. 

https://reference.opcfoundation.org/v104/Core/docs/Part19/7.1/ defines the MultiStateDictionaryEntryDiscreteBaseType as Scalar.
The xml omit it, as scalar is the default value. As https://reference.opcfoundation.org/v104/Core/docs/Part3/6.2.7/ defines, override is not possible, so it's not valid to set this to "-2" by PADIM.

From: https://reference.opcfoundation.org/v104/Core/docs/Part6/F.8/
>  ValueRank: The value rank. If not specified, the default value is -1 (Scalar)

I would suggest to add an Mantis issue at the OPC Foundation:
https://apps.opcfoundation.org/mantis/

PS: Sorry for the many hyperlinks, but I have tried to reproduce the problem according to the specification and document this for further discussions.

mgroenhoff, 2020-08-06:
I think you mean that it is an issue with the core Schema instead because MultiStateDictionaryEntryDiscreteBaseType is defined there and not in PADIM.

Thanks for your help. I will create an issue there.

mgroenhoff, 2020-08-06:
If anyone is running into the same issue. I have created a workaround while the issue is unresolved:

```c
// Instead of this
// server = UA_Server_new();

// I now temporarily use this
UA_ServerConfig config;
memset(&config, 0, sizeof(UA_ServerConfig));
config.logger = UA_Log_Stdout_;
UA_Nodestore_HashMap(&config.nodestore);
// Workaround for https://apps.opcfoundation.org/mantis/view.php?id=5862
config.nodeLifecycle.constructor = [](UA_Server* server, const UA_NodeId* sessionId, void* sessionContext, const UA_NodeId* nodeId, void** nodeContext) -> UA_StatusCode {
    auto fixNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_MULTISTATEDICTIONARYENTRYDISCRETEBASETYPE);
    auto config =  UA_Server_getConfig(server);
    auto fixNode = config->nodestore.getNode(config->nodestore.context, &fixNodeId);
    if (fixNode) {
        const_cast<UA_VariableTypeNode*>(reinterpret_cast<const UA_VariableTypeNode*>(fixNode))->valueRank = -2;
        config->nodestore.releaseNode(config->nodestore.context, fixNode);
        config->nodeLifecycle.constructor = nullptr;
    }
    return UA_STATUSCODE_GOOD;
};
server = UA_Server_newWithConfig(&config);
```

ccvca, 2020-08-06:
> I think you mean that it is an issue with the core Schema instead because MultiStateDictionaryEntryDiscreteBaseType is defined there and not in PADIM.

This depends on the view, if you exptect "MultiStateDictionaryEntryDiscreteBaseType" to be right, then ValueRank of MultiStateDiscreteSignalVariableType is wrong.

mgroenhoff, 2020-08-06:
True but everything "up" the type hierarchy (`BaseVariableType -> BaseDataVariableType -> DataItemType -> DiscreteItemType -> MultiStateValueDiscreteType -> MultiStateDictionaryEntryDiscreteBaseType`) also define their ValueRank to be -2 and as you mentioned, subtypes can not have a different ValueRank than it its basetype. To me it seems that the error is with `MultiStateDictionaryEntryDiscreteBaseType` not specifying its ValueRank to be -2 or maybe I don't fully understand it yet.

teccheck, 2025-05-21:
As commented, it seems open62541 works according to spec and there's a problem with the nodeset, so I'll close this issue

closed 2025-05-21 by teccheck