open62541/open62541 #3833: Maximum client connection

kusumy, 2020-08-07:
## Description

How many maximum client can be connected to a OPC UA server in open62541 SDK? 

assumsing all client are active.

Regards

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-08-13:
This can be configured in the ServerConfig strict.
If you set the max to zero, then there is no limit.

NoelGraf, 2021-04-30:
I would then close the issue.

closed 2021-04-30 by NoelGraf

kusumy, 2021-05-20:
> This can be configured in the ServerConfig strict.
> If you set the max to zero, then there is no limit.

Should i set maxSecureChannels or maxSessions ?

Regards