open62541/open62541 #3840: Support for the OPC UA 1.04 profiles

daab007, 2020-08-13:
Hello,

you use deprecated names for the profiles when you talk about certification. Does it mean that OPC UA 1.04 profiles were not tested? Could we expect any problems with that?

It could be a potential showstopper for us...

Best Regards,
Daniel Cyran.

jpfr, 2020-08-13:
The 1.04 profiles were not used by the testing lab one year ago.

We are now using the current version of the CTT internally.

jpfr, 2020-08-13:
Note that the release family 1.0 is certified.
Certification of new features since 1.0 is not yet done.

daab007, 2020-08-13:
It is clear, that not all newest features were certified in the past.

I am wondering about potential problems with the features/facets  already certified in the past in regards to new profiles.  
What is more we are planning to introduce Alarm and Condition Server as well. Same question here... Any advises here?

daab007, 2020-08-13:
Thanks for the feedback, will contact you later on.

closed 2020-08-13 by daab007