open62541/open62541 #3844: "Sending OPN message failed with error BadInvalidArgument" with OpenSSL and private key in PEM format.

basyskom-jvoe, 2020-08-14:
## Description
If a PEM encoded private key is used with the OpenSSL based security plugins, the initialization is successful, but the connect fails with ```Sending OPN message failed with error BadInvalidArgument```.

The same combination of certificate and private key works with the mbedTLS based security plugins.

## Background Information / Reproduction Steps
Used CMake options:

```bash
cmake -DUA_ENABLE_AMALGAMATION -DUA_ENABLE_ENCRYPTION -DUA_ENABLE_ENCRYPTION_OPENSSL.
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-08-14:
> Edit: UA_ClientConfig_setDefaultEncryption() from the OpenSSL plugin fails with UA_STATUSCODE_BADINTERNALERROR if a PEM encoded CRL is used.

Maybe we are just missing a log output at that location.

basyskom-jvoe, 2020-08-14:
Similar problem for the trust list: I have two trusted certificates in DER format.
With OpenSSL, UA_ClientConfig_setDefaultEncryption() returns UA_STATUSCODE_GOOD, but the connect fails with the following message:
```
[2020-08-14 12:36:07.813 (UTC+0200)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2020-08-14 12:36:07.813 (UTC+0200)] info/securitypolicy	The Basic128Rsa15 security policy channel with openssl is created.
[2020-08-14 12:36:07.918 (UTC+0200)] warn/channel	Connection 6 | SecureChannel 0 | Could not verify the server's certificate
[2020-08-14 12:36:07.918 (UTC+0200)] info/securitypolicy	The Basic128Rsa15 security policy channel with openssl is deleted.
```
The same code with the same trusted certificates works with the mbedTLS plugin.

basyskom-jvoe, 2020-08-19:
PR with a fix: #3846

jpfr, 2020-08-19:
Fix merged.
Thanks @basyskom-jvoe

closed 2020-08-19 by jpfr