open62541/open62541 #3849: OPCUA PubSub over AMQP

waheedejaz, 2020-08-18:
## Description
Add support for AMQP transport
## Background Information / Reproduction Steps
Using Qpid Proton lib c
https://github.com/apache/qpid-proton/tree/master/c

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

waheedejaz, 2020-08-18:
Please give an early review. The AMQP publish example works with broker example in qpid proton lib proton/c/examples/broker.

The build system is temporarily modified to link the proton lib as it doesn't build successfully as an object library owing to all the warnings enabled in the open62541 build system.

jackybek, 2021-04-23:
Hi,
I am trying to evaluate .. but when i download the qpid-proton-cpp tar file, there are only 2 files. 
a. libqpid-proton-cpp.so.12.0.0 
b. libqpid-proton-cpp.so.12 

Seemed that libqpid-proton-cpp.a is missing.
I am getting the following link errors:

...
/usr/bin/ld: //usr/local/lib/libqpid-proton-cpp.so: undefined reference to `pn_data_get_binary'
/usr/bin/ld: //usr/local/lib/libqpid-proton-cpp.so: undefined reference to `pni_message_get_extra'
/usr/bin/ld: //usr/local/lib/libqpid-proton-cpp.so: undefined reference to `pn_condition_copy'
/usr/bin/ld: //usr/local/lib/libqpid-proton-cpp.so: undefined reference to `pn_message_set_content_type'
/usr/bin/ld: //usr/local/lib/libqpid-proton-cpp.so: undefined reference to `pn_message_is_durable'
/usr/bin/ld: //usr/local/lib/libqpid-proton-cpp.so: undefined reference to `pn_message_get_correlation_id'
/usr/bin/ld: //usr/local/lib/libqpid-proton-cpp.so: undefined reference to `pn_ssl_domain_free'
...

Seemed that it is looking for proton lib functions..

During the installation process, i already did the following:
1. sudo lfconfig -v
2. update /etc/environment to include the path
3. source /etc/environment and confirm that the libraries are loaded

The only thing i suspect is the missing <.a> file that normally comes together with the <.so> file ... can you advise?
Thanks.

jackybek, 2021-04-25:
solve partially by indicating -Lqpid-proton-cpp in gcc
the errors are reduced to :

/usr/bin/ld: /tmp/ccXTNbuu.o: in function `_write':
/home/pi/OPCUAProject/ua_amqp_adaptor.c:32: undefined reference to `pn_connection_driver_write_buffer'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:42: undefined reference to `pn_connection_driver_write_done'
/usr/bin/ld: /tmp/ccXTNbuu.o: in function `_read':
/home/pi/OPCUAProject/ua_amqp_adaptor.c:59: undefined reference to `pn_connection_driver_read_buffer'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:67: undefined reference to `pn_connection_driver_read_done'
/usr/bin/ld: /tmp/ccXTNbuu.o: in function `UA_AmqpDisconnect':
/home/pi/OPCUAProject/ua_amqp_adaptor.c:81: undefined reference to `pn_connection_close'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:84: undefined reference to `pn_message_error'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:84: undefined reference to `pn_error_text'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:87: undefined reference to `pn_collector_free'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:88: undefined reference to `pn_connection_driver_close'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:89: undefined reference to `pn_connection_driver_destroy'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:92: undefined reference to `pn_message_free'
/usr/bin/ld: /tmp/ccXTNbuu.o: in function `UA_AmqpConnect':
/home/pi/OPCUAProject/ua_amqp_adaptor.c:107: undefined reference to `pn_connection_driver_init'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:112: undefined reference to `pn_connection_open'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:115: undefined reference to `pn_session'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:116: undefined reference to `pn_session_open'
/usr/bin/ld: /tmp/ccXTNbuu.o: in function `publishAmqp':
/home/pi/OPCUAProject/ua_amqp_adaptor.c:168: undefined reference to `pn_sender'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:172: undefined reference to `pn_link_target'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:172: undefined reference to `pn_terminus_set_address'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:173: undefined reference to `pn_link_open'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:175: undefined reference to `pn_message'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:192: undefined reference to `pn_link_credit'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:199: undefined reference to `pn_dtag'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:199: undefined reference to `pn_delivery'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:204: undefined reference to `pn_message_clear'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:205: undefined reference to `pn_message_body'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:207: undefined reference to `pn_message_id'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:207: undefined reference to `pn_data_put_int'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:208: undefined reference to `pn_data_put_map'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:209: undefined reference to `pn_data_enter'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:210: undefined reference to `pn_bytes'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:210: undefined reference to `pn_data_put_string'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:211: undefined reference to `pn_data_put_int'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:212: undefined reference to `pn_data_exit'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:213: undefined reference to `pn_message_send'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:216: undefined reference to `pn_message_error'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:216: undefined reference to `pn_error_text'
/usr/bin/ld: /tmp/ccXTNbuu.o: in function `__handle':
/home/pi/OPCUAProject/ua_amqp_adaptor.c:251: undefined reference to `pn_event_type'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:258: undefined reference to `pn_event_link'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:260: undefined reference to `pn_link_is_sender'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:262: undefined reference to `pn_event_delivery'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:263: undefined reference to `pn_delivery_remote_state'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:267: undefined reference to `pn_event_delivery'
/usr/bin/ld: /tmp/ccXTNbuu.o: in function `yieldAmqp':
/home/pi/OPCUAProject/ua_amqp_adaptor.c:303: undefined reference to `pn_connection_driver_next_event'
/usr/bin/ld: /home/pi/OPCUAProject/ua_amqp_adaptor.c:304: undefined reference to `pn_connection_driver_next_event'
collect2: error: ld returned 1 exit status

closed 2025-05-21 by teccheck