open62541/open62541 #3853: Memory not properly handled in UA_Server_newWithConfig

cochicde, 2020-08-20:
## Description

Memory not properly handled in UA_Server_newWithConfig

## Background Information / Reproduction Steps

While trying to fix issues regarding malloc failures, I found an issue in this function

https://github.com/open62541/open62541/blob/ee275e79310280a71fad3b21f6d430c0c046091b/src/server/ua_server.c#L341-L350

The problem is that the parameter config has some allocated memory in it (in the hashMap). Now, this function can fail in two places:
 - UA_calloc (line 345) wich returns NULL. The allocated memory should be deallocated by the one who called this function
 - UA_Server_init (Line 350): If this fails, the UA_Server_init will take care of deleteing the whole server, including the its config variable, which was COPIED in line 349. That means that the function will return NULL, but the memory in this case is already deallocated.

What happens now, is that the function that called UA_Server_newWithConfig can receive a NULL and won't know if the memory was deallocated already or not. 

My first guess was that the caller can check the parameter he send, and see if the memory pointers are NULL. BUT, since the config variable was copied, the copied config variable in the server might have a NULL pointer in it (failure case 2), but the config parameter varaible will never be NULL at this point, since it has never changed (the pointer variable in it is still non-NULL, although the memory where is pointing was already deallocated)

I tried to clean the config parameter variable in UA_Server_newWithConfig but it's const.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-08-20:
Good find.

The Server_new function effectively *moves* the config into the server.
We could change the default behaviour to always zero-out the content of the input UA_ServerConfig before returning.

Then it is only logical that we clean up the dynamically allocated memory in the config if something fails during initialisation.

cochicde, 2020-08-20:
> 
> 
> Good find.
> 
> The Server_new function effectively _moves_ the config into the server.

Did you mean UA_Server_newWithConfig?

> We could change the default behaviour to always zero-out the content of the input UA_ServerConfig before returning.
> 
> Then it is only logical that we clean up the dynamically allocated memory in the config if something fails during initialisation.

I didn't understand where exactly to do what. The UA_ServerConfig parameter is const, so the only thing I can think of is, if calloc fails, config should be move to a dummy local variable wihch will be used to call UA_ServerConfig_clean. Something like this:

```
UA_Server *
UA_Server_newWithConfig(const UA_ServerConfig *config) {
    if(!config)
        return NULL;
    UA_Server *server = (UA_Server *)UA_calloc(1, sizeof(UA_Server));
     if(!server) { 
       UA_ServerConfig dummyConfig = *config;
       UA_ServerConfig_clean(&dummyConfig);
       return NULL;
    }
    server->config = *config;
    return UA_Server_init(server);
}
```

am I missing something?

jpfr, 2020-08-20:
Or make the argument non-const if we reset the struct to all zeros after moving the content.

cochicde, 2020-08-21:
wouldn't that be a change in the API?

PR #3855 (2020-08-24): Fix leak in UA_Server_newWithConfig
Fix #3853
closed 2020-08-24 by jpfr