open62541/open62541 #3857: ServerNetworkLayerTCP_start() fails completely if IPv6 is not available

tboeckel, 2020-08-26:
## Description
The function ServerNetworkLayerTCP_start() in arch/network_tcp.c fails completely if the socket cannot be opened for at least one addrinfo entry, even if the first entry succeeded.

## Background Information / Reproduction Steps
The commit a97cdb31b8a7c5ca429bf90b55db9360e8b56c9f introduced the additional check for failure.

It seems this issue appears on certain systems only. It seems to be related to the availablility of IPv6 on a network interface. On one system running an older Linaro Linux version I have this eth0 network interface without IPv6 capability:

eth0      Link encap:Ethernet  HWaddr fc:c2:3d:18:31:37
          inet addr:192.168.100.22  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:15261 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9298 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:11549949 (11.5 MB)  TX bytes:1015216 (1.0 MB)
          Interrupt:54 Base address:0xb000 

On a more recent Yocto Linux system I have this eth0 network interface with IPv6 capability:

eth0      Link encap:Ethernet  HWaddr fc:c2:3d:15:61:8b  
          inet addr:192.168.100.21  Bcast:192.168.100.255  Mask:255.255.255.0
          inet6 addr: fe80::fec2:3dff:fe15:618b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:51 errors:0 dropped:0 overruns:0 frame:0
          TX packets:67 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5724 (5.5 KiB)  TX bytes:9715 (9.4 KiB)
          Interrupt:32 Base address:0xb000 

On the first system (Linaro Linux) the UA_socket() call in addServerSocket() fails with the error message "Error opening the server socket: Address family not supported by protocol" while ai_family in the addrinfo structure is AF_INET6. The first call of addServerSocket() with ai_family=AF_INET succeeded before.

On the second system (Yocto Linux) the UA_socket() call never fails for any addrinfo entry. This is obvious, because this system has IPv6 capabilities.

Release 1.1.1 of open62541 did not check the success of addServerSocket() and hence never failed obviously, although the UA_socket() call must have failed as well, but without any error message and without any implied failure.

My current workaround is to comment out this check for success of addServerSocket(), but I would like to remove this private patch again as soon as possible.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Linaro Linux, Yocto Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

daab007, 2020-10-19:
Any update? Got a same problem... Seems like related to: #3810

tboeckel, 2022-04-28:
For me this issue is fixed since the release of 1.2.3.

closed 2022-04-28 by tboeckel