open62541/open62541 #3860: Custom Structure and Enum

wentabat, 2020-08-27:
## Description

## Background Information / Reproduction Steps

These 2 issues are related, 1 each for Structure and Enum.
The client used is the UaExpert.

### 1) Structure
​
In the Data Access View, there are two variables.   
 - var-struct - data type is 'MyStruct', a custom data type structure created on "RUNTIME" on the heap for its UA_DataType and UA_DataTypeMember for its members. "Not generated from command line"
 - var-BuildInfo - data type is 'BuildInfo' built-in type from namespace 0.
In the DataType column, var-struct is showing "ExtensionObject" while var-BuildInfo is showing "ExtensionObject - BuildInfo"

Although the DataType in the Attributes window shows MyStruct1 node details. the client is not able to fully decode the DataType of var-struct, instead it just display the generic 'ExtensionObject' as opposed to the 'ExtensionObject-BuildInfo'
As a result, when you double click the value, the popup window value is not able to display the fields of the struct. See screenshots below for both comparisons.

MyStruct1: node id=17189 namespace=1 (User namespace)
[image]
[image]
[image]

**V.S.**

BuildInfo: node id=338   namespace=0 (OPC UA)
[image]
[image]

### 2) Enumeration
The issue for enumeration is somewhat similar to the Structure. The same process in creating the Enum on runtime using UA_DataType and UA_DataTypeMember .
- var-enum- data type is 'MyEnum', a custom data type enum.  Dropdown list is not showing.
- var-enumIdType- data type is 'IdType' built-in type from the namespace 0 node id:256.  Dropdown list is shown.

[image]

Any help would very much appreciated.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.1.2
 - [X] Other OPC UA SDKs used (client or server):  UaExpert
 - [X] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ZidCode, 2021-07-28:
Hi, 
could you give more infos? How you modelled it etc... I got a similiar issue, that UAexpert recognizes the datatype but"writing" to it causes a BadTypeMismatch.

Cheers

teccheck, 2025-05-21:
As we don't have any further information and there's been no recent activity, this issue seems abandoned. Feel free to reopen if needed

closed 2025-05-21 by teccheck