open62541/open62541 #3862: OPCUA PUBSUB over TSN: ingress,Cannot find specified qdisc on specified device.

xuexiqiangzhe, 2020-08-28:
Hello
My system is ubuntu 16.04
The kernel is 4.19.37-rt19.
I follow example/pubsub_realtime/README_pubsub_TSN.txt of master branch.
I have enable sch_etf,sch_mqprio,8021q.
No problem until following line 85,error
```
$ sudo tc qdisc delete dev enp1s0 ingress
Error: Cannot find specified qdisc on specified device.
```
Can anyone help me,thanks!

opcua-tsn-team-kalycito, 2020-08-28:
Can you provide some additional information,

- Hardware that you are using? (The whole setup is tested in i210 hardware)

- Installed version of iproute2? (iproute2 4.19 is recommended)

- tc statistics by using this command "tc -s qdisc show dev enp1s0" (at two points: before initial configuration and before this error occurs)

xuexiqiangzhe, 2020-08-28:
Hi,
Both nodes:I210 hardware.
iproute2   5.8.0v.
```
tc -s qdisc show dev enp1s0
qdisc mq 0: root 
 Sent 9812 bytes 54 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc pfifo_fast 0: parent :4 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
 Sent 1036 bytes 10 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc pfifo_fast 0: parent :3 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
 Sent 1320 bytes 12 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc pfifo_fast 0: parent :2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
 Sent 2326 bytes 17 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc pfifo_fast 0: parent :1 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
 Sent 5130 bytes 15 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
```
thank you!

opcua-tsn-team-kalycito, 2020-08-31:
Can you check with the current master with the updated commands?

If the issue persists, try with iproute2 4.19 version.

NoelGraf, 2021-04-29:
@xuexiqiangzhe have you tried it again?

teccheck, 2025-05-21:
Because there's been no recent activity, I'll close this

closed 2025-05-21 by teccheck