open62541/open62541 #3870: UA_Client_readValueAttribute interval

angrybinary, 2020-09-02:
## Description

On client side I try to read a node: 
```
	while (clientRunning) {

		status = UA_Client_readValueAttribute(client, UA_NODEID_STRING(1, (char *)"test.tag"), &value);

		if (status == UA_STATUSCODE_GOOD &&
			UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_INT16])) {

			print_time();
			printf("Node value: %i\n\r", *(UA_Int16*)value.data);
		}
		Sleep(5);
	}
```
On server side I have following:

```
    config->publishingIntervalLimits = {3.0, 5.0};
    config->samplingIntervalLimits = {3.0, 5.0};
```
And incrementing node value every 10 ms.

In client log I get:
```
[2020-09-02 18:28:57.463 (UTC+0200)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-09-02 18:28:57.590 (UTC+0200)] info/channel	Connection 408 | SecureChannel 5 | Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-09-02 18:28:57.590 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2020-09-02 18:28:57.639 (UTC+0200)] info/client	Selected Endpoint opc.tcp://PC:5333 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-09-02 18:28:57.639 (UTC+0200)] info/client	Selected UserTokenPolicy open62541-anonymous-policy with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-09-02 18:28:57.689 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2020-09-02 18:28:57.739 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2020/09/02 18:28:57.789] Node value: 6600
[2020/09/02 18:28:57.839] Node value: 6605
[2020/09/02 18:28:57.889] Node value: 6609
[2020/09/02 18:28:57.940] Node value: 6614
[2020/09/02 18:28:57.990] Node value: 6619
[2020/09/02 18:28:58.040] Node value: 6623
[2020/09/02 18:28:58.090] Node value: 6628
[2020/09/02 18:28:58.139] Node value: 6633
[2020/09/02 18:28:58.189] Node value: 6637
[2020/09/02 18:28:58.239] Node value: 6642
[2020/09/02 18:28:58.289] Node value: 6646
[2020/09/02 18:28:58.339] Node value: 6651
```
Node value is read every 50 ms. 
How can I control/increase read interval?

## Background Information / Reproduction Steps

Used CMake options:

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1-44-gb760b48e
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Win
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

heppth, 2020-09-03:
I am relatively sure that it is not possible to delay the response time of readValueAttribute() calls.
If your client subscribes the node via monitored item, the server is able to control/increase the interval.
This would be the common way to get cyclical value changes.

angrybinary, 2020-09-05:
On server side I write value every 10 ms
```
	UA_Variant value;
	UA_NodeId NodeId = UA_NODEID_STRING(1, (char *)"test.tag");

	short number = 0;
	std::thread serverThread(startServer);

	while (serverRunning) {

		UA_Variant_setScalar(&value, &number, &UA_TYPES[UA_TYPES_INT16]);

		status = UA_Server_writeValue(server, NodeId, value);

		if (status == UA_STATUSCODE_GOOD ) {

			print_time();
			printf("Value writen: %i\n\r", *(UA_Int16*)value.data);
		}

		number++;
		Sleep(10);
	}
```
But on a client side I see change only every 50 ms. 

Server:
```
[2020/09/05 19:07:56.503] Value writen: 148
[2020/09/05 19:07:56.514] Value writen: 149
[2020/09/05 19:07:56.524] Value writen: 150
[2020/09/05 19:07:56.535] Value writen: 151
[2020/09/05 19:07:56.545] Value writen: 152
[2020/09/05 19:07:56.555] Value writen: 153
[2020/09/05 19:07:56.566] Value writen: 154
[2020/09/05 19:07:56.576] Value writen: 155
[2020/09/05 19:07:56.587] Value writen: 156
[2020/09/05 19:07:56.598] Value writen: 157
[2020/09/05 19:07:56.608] Value writen: 158
[2020/09/05 19:07:56.618] Value writen: 159
[2020/09/05 19:07:56.629] Value writen: 160
```

Client:
```
[2020/09/05 19:07:56.508] Node value: 148
[2020/09/05 19:07:56.557] Node value: 153
[2020/09/05 19:07:56.607] Node value: 157
[2020/09/05 19:07:56.657] Node value: 162
[2020/09/05 19:07:56.707] Node value: 167
```
Does this mean that even if I write value every 10 ms it is not renewed that frequently?

angrybinary, 2020-09-25:
Was setting sampling interval at the wrong moment and it used to get overwritten.

closed 2020-09-25 by angrybinary