open62541/open62541 #3873: Bug causes 100% CPU in OPC-UA client if server refuses connection

heppth, 2020-09-07:
After updating from version 0.3 to version 1.1 it often happens that the application has 100% CPU load. After a longer analysis the cause seems to be found. An `UA_sleep_ms(100)` was removed during a refactoring between line 804 and 805. The comment for the `UA_sleep_ms(100)` is still there.

https://github.com/open62541/open62541/blob/40cb2f0e46ccdb04ecfe4fcf5a133257aeed3ab1/arch/network_tcp.c#L783-L805

Here is the code from version 0.3. You can see there was an `UA_sleep_ms(100)` in case of `so_error == ECONNREFUSED`.

https://github.com/open62541/open62541/blob/f93e695f029297938d7370a3b6e2e17df0aaf378/plugins/ua_network_tcp.c#L915-L934

I thought about fixing this bug, but I do not understand why an exception is made for `so_error == ECONNREFUSED`. If the server refuses the connection, any further call to select() does not make sense to me. Just like with any other error, we should simply disconnect and return `UA_STATUSCODE_BADDISCONNECT`.

jpfr, 2020-09-09:
Hmm. This is covered by the unit tests if I recall correctly.
Let us look into that.

If the network layer aborts the connection at this level, the client should set its "connectStatus" to the error code and abort.

Are you calling UA_Client_connect in a loop?
Or UA_Client_connect just once and thereafter UA_Client_run_iterate?

We might be more precise in the docs.

heppth, 2020-09-09:
> Are you calling UA_Client_connect in a loop?
> Or UA_Client_connect just once and thereafter UA_Client_run_iterate?

I am calling UA_Client_connect just once and thereafter UA_Client_run_iterate

heppth, 2020-09-11:
I found another bug in `UA_ClientConnectionTCP_poll`, which can have similar effects.
https://github.com/open62541/open62541/blob/40cb2f0e46ccdb04ecfe4fcf5a133257aeed3ab1/arch/network_tcp.c#L772
If select() returns 0 (timeout), then the socketfd will never be closed.
If you try to open the connection over and over again there will be no more file descriptors available.

jpfr, 2020-10-03:
Your second comment is addressed in

https://github.com/open62541/open62541/pull/3930/commits/e9f59b7688d07d00de9b6b5b7023895d97a37d45

teccheck, 2025-05-21:
The relevant code gone because of the new event loop which brought a complete overhaul, so I'll close this

closed 2025-05-21 by teccheck

paddor, 2026-01-16:
I'm still getting 100% from UA client during exit about 1/3rd of the time. Using `UA_Client_connect`  and `UA_Client_run_iterate`. Debugger shows nothing and `/proc/PID/stack` is empty at that time. How can I debug this?