open62541/open62541 #3878: historical data with UA_Server_addDataSourceVariableNode

viktor-morin, 2020-09-09:
- [x] open62541 Version (release number or git tag): (2d2b98b)
 - [x] Other OPC UA SDKs used (client or server): None
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

I have tested to compile and use follwing example:
[tutorial_server_historicaldata.c](https://github.com/open62541/open62541/blob/master/examples/tutorial_server_historicaldata.c)

I would like to use 
```c
UA_Server_addDataSourceVariableNode
```
instead of 
```c
UA_Server_addVariableNode
```

The results is:

> No history values received for node 'NS1|Numeric|1337'

**Any suggestions what the difference is, that i need to handle?**

Example code:
```c
#include <open62541/plugin/historydata/history_data_backend_memory.h>
#include <open62541/plugin/historydata/history_data_gathering_default.h>
#include <open62541/plugin/historydata/history_database_default.h>
#include <open62541/plugin/historydatabase.h>
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

#define LOCALE "en-US"
#define VECTORSIZE 1000

// TEMP
typedef struct
{
   double doubleValue;
   bool boolValue;
   float floatArray[1000];
} TestClass;
// TEMP

typedef struct
{
   UA_UInt16 nameSpace;
   int nodeId;
   char *nodeName;
   UA_UInt16 parentNameSpace;
   int parentNodeId;
} objectNodesStruct;

typedef struct
{
   UA_UInt16 nameSpace;
   int nodeId;
   char *nodeName;
   int nodeType;
   UA_Byte nodeAccess;
   UA_UInt16 parentNameSpace;
   int parentNodeId;
   int *dataPointer;
[... 170 lines cut ...]

   int i;
   int objectArrayLength = sizeof(objectNodes) / sizeof(objectNodes[0]);
   for (i = 0; i < objectArrayLength; i++)
      addObjectNode(server, objectNodes[i]);

   int variableArrayLength = sizeof(variableNodes) / sizeof(variableNodes[0]);
   for (i = 0; i < variableArrayLength; i++)
      addDataSourceVariableNode(server, setting, gathering, variableNodes[i]);

   while (running)
   {
      UA_Server_run_iterate(server, waitInternal);
      // printf("value in memory: %f \n", testClass.doubleValue);
      // printf("value in memory: %d \n", testClass.boolValue);
      // usleep(10000);
   }

   retval = UA_Server_run_shutdown(server);

cleanup:
   UA_Server_delete(server);
   return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

viktor-morin, 2020-09-10:
Updated example code to only contains necessary

viktor-morin, 2020-09-15:
After more investigation my best guess is following:

**Setup of the node using addDataSourceVariableNode**

```c
    UA_DataSource dataSource;
    dataSource.read = readDouble;
    dataSource.write = writeDouble;
    UA_Server_addDataSourceVariableNode(server, currentNodeId, parentNodeId,
                                        parentReferenceNodeId, nodeName,
                                        variableTypeNodeId, attr, dataSource, (int *)&testClass.doubleValue, &outNodeId);
```

**Implementaiton of read/write for datatype UA_Double**
```c
static UA_StatusCode
readDouble(UA_Server *server,
           const UA_NodeId *sessionId, void *sessionContext,
           const UA_NodeId *nodeId, void *nodeContext,
           UA_Boolean sourceTimeStamp, const UA_NumericRange *range,
           UA_DataValue *dataValue)
{
    UA_Double memoryValue = *(double *)nodeContext;
    UA_Variant_setScalarCopy(&dataValue->value, &memoryValue,
                             &UA_TYPES[UA_TYPES_DOUBLE]);
    dataValue->hasValue = true;
    return UA_STATUSCODE_GOOD;
}

static UA_StatusCode
writeDouble(UA_Server *server,
            const UA_NodeId *sessionId, void *sessionContext,
            const UA_NodeId *nodeId, void *nodeContext,
            const UA_NumericRange *range, const UA_DataValue *data)
{
    *(double *)nodeContext = *((double *)data->value.data);
    return UA_STATUSCODE_GOOD;
}
```

I assume the issue is that my function 

> writeDouble

**dosn't actually write to any history database?** 

>

juangburgos, 2020-10-09:
I found out a while ago that if you use data sources, the history service does not get called, meaning you have to call it yourself. Why? It beats me, take a look [at the source code of `writeValueAttribute` in `src/server/ua_services_attribute.c`](https://github.com/open62541/open62541/blob/52db1ca7a44758c86ec3df00945f5c03771eb6dd/src/server/ua_services_attribute.c#L1285):

```c
static UA_StatusCode
writeValueAttribute(UA_Server *server, UA_Session *session,
                UA_VariableNode *node, const UA_DataValue *value,
                const UA_String *indexRange) {

// Some stuff...

switch(node->valueBackend.backendType) {
    case UA_VALUEBACKENDTYPE_NONE:
        /* Ok, do it */
        if(node->valueSource == UA_VALUESOURCE_DATA) {
            /* Set the source timestamp if there is none */
            UA_DateTime now = UA_DateTime_now();
            if(!adjustedValue.hasSourceTimestamp) {
                adjustedValue.sourceTimestamp = now;
                adjustedValue.hasSourceTimestamp = true;
            }

            if(!adjustedValue.hasServerTimestamp) {
                adjustedValue.serverTimestamp = now;
                adjustedValue.hasServerTimestamp = true;
            }

            if(!rangeptr)
                retval = writeValueAttributeWithoutRange(node, &adjustedValue);
            else
                retval = writeValueAttributeWithRange(node, &adjustedValue, rangeptr);

#ifdef UA_ENABLE_HISTORIZING
            /* node is a UA_VariableNode*, but it may also point to a UA_VariableTypeNode */
    /* UA_VariableTypeNode doesn't have the historizing attribute */
    if(retval == UA_STATUSCODE_GOOD &&
       node->head.nodeClass == UA_NODECLASS_VARIABLE &&
       server->config.historyDatabase.setValue) {
        UA_UNLOCK(server->serviceMutex);
        server->config.historyDatabase.
            setValue(server, server->config.historyDatabase.context,
                     &session->sessionId, session->sessionHandle,
                     &node->head.nodeId, node->historizing, &adjustedValue);
        UA_LOCK(server->serviceMutex);
    }
#endif
            /* Callback after writing */
            if(retval == UA_STATUSCODE_GOOD && node->value.data.callback.onWrite) {
                UA_UNLOCK(server->serviceMutex)
                node->value.data.callback.
                    onWrite(server, &session->sessionId, session->sessionHandle,
                            &node->head.nodeId, node->head.context, rangeptr, &adjustedValue);
                UA_LOCK(server->serviceMutex);

            }
        } else {
            if(node->value.dataSource.write) {
                UA_UNLOCK(server->serviceMutex);
                retval = node->value.dataSource.
                    write(server, &session->sessionId, session->sessionHandle,
                          &node->head.nodeId, node->head.context, rangeptr, &adjustedValue);
                UA_LOCK(server->serviceMutex);
            } else {
                retval = UA_STATUSCODE_BADWRITENOTSUPPORTED;
            }
        }
        break;
    case UA_VALUEBACKENDTYPE_INTERNAL:
        break;
    case UA_VALUEBACKENDTYPE_DATA_SOURCE_CALLBACK:
        break;
    case UA_VALUEBACKENDTYPE_EXTERNAL:
        if(node->valueBackend.backend.external.callback.userWrite == NULL){
            if(rangeptr)
                UA_free(range.dimensions);
            return UA_STATUSCODE_BADWRITENOTSUPPORTED;
        }
        node->valueBackend.backend.external.callback.userWrite(server, &session->sessionId, session->sessionHandle,
                                                             &node->head.nodeId, node->head.context, rangeptr, &adjustedValue);
        break;
}
```

The `server->config.historyDatabase.setValue` only gets called if `node->valueBackend.backendType == UA_VALUEBACKENDTYPE_NONE` **and** `node->valueSource == UA_VALUESOURCE_DATA`, if it is `UA_VALUESOURCE_DATASOURCE` you are on your own.

viktor-morin, 2020-10-13:
@jpfr 
Is this a bug that's need to be fixed or is there another way to handle this?

jackybek, 2020-12-08:
Hi Viktor 
it is nice to know that you are able to use historizing features.  Can I know what version of gnu compiler and open62641 package you are using ?  And did you turn on alamagation during the make process?
Reason is I tried the same code but it fail to compile.
Many thanks for your advice.

viktor-morin, 2020-12-09:
> Hi Viktor
> it is nice to know that you are able to use historizing features. Can I know what version of gnu compiler and open62641 package you are using ? And did you turn on alamagation during the make process?
> Reason is I tried the same code but it fail to compile.
> Many thanks for your advice.
@jackybek I havn't been able to use it together with datasource. As mention above by @juangburgos it dosnt work, or atleast didnt work at the time. Havn't tried with the latest version today.

viktor-morin, 2022-06-08:
Not working with this anymore, no information regarding if this is fixed or not.

closed 2022-06-08 by viktor-morin