open62541/open62541 #3879: BadServiceUnsupported during call RegisterServer to OPC UA Foundation LDS

mateusz-talma, 2020-09-09:
Hello,

I'm implementing an OPC UA server. I have a problem with register my opcua server to OPC UA Foundation LDS v1.04.402.447 (https://apps.opcfoundation.org/LDS/). I run discovery server with default ini file. I use self-signed certificate, the same for server and clients (required for register in discovery). This certificate is on the OPCUA Foundation LDS trusted list. 

  I have tried two methods: 

## 1. UA_Server_addPeriodicServerRegisterCallback

```cpp
  m_client = UA_Client_new();
  UA_ClientConfig *m_clientConfig = UA_Client_getConfig(m_client);
  UA_ClientConfig_setDefaultEncryption(m_clientConfig, m_certificate, m_privateKey, nullptr, 0, nullptr, 0);

  m_clientConfig->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;

  m_clientConfig->clientDescription.applicationUri = UA_String_fromChars(theApp.GetAppUri().c_str());
  m_clientConfig->clientDescription.productUri = UA_STRING_ALLOC("xxx");
  m_clientConfig->clientDescription.applicationName = UA_LOCALIZEDTEXT_ALLOC("pl-pl", "xxx");
  m_clientConfig->clientDescription.applicationType = UA_APPLICATIONTYPE_SERVER;
  m_clientConfig->clientDescription.applicationUri = UA_String_fromChars(theApp.GetAppUri().c_str());
  m_clientConfig->logger.log = AddDiagnosticLog;
  m_clientConfig->securityPolicyUri = UA_STRING("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15");

  auto status = UA_Server_addPeriodicServerRegisterCallback(
    m_serverUA, m_client, "opc.tcp://localhost:4840", 10 * 60 * 1000, 500, nullptr);
```

Results:
Log open62541
```
2020-09-09 12:37.07 info/network TCP network layer listening on opc.tcp://xxx:55555/
2020-09-09 12:37.08 warn/client The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
2020-09-09 12:37.08 trace/client Client connect iterate
2020-09-09 12:37.08 trace/client Client connect iterate
2020-09-09 12:37.08 debug/network Sent HEL message
2020-09-09 12:37.08 debug/client Client Status: ChannelState: HELSent, SessionState: Closed, ConnectStatus: Good
2020-09-09 12:37.08 trace/client Client connect iterate
2020-09-09 12:37.08 debug/network Received ACK message
2020-09-09 12:37.08 debug/client Client Status: ChannelState: AckReceived, SessionState: Closed, ConnectStatus: Good
2020-09-09 12:37.08 trace/client Client connect iterate
2020-09-09 12:37.08 debug/channel Connection 1308 | SecureChannel 0 | Generating new local nonce
2020-09-09 12:37.08 debug/channel Connection 1308 | SecureChannel 0 | Requesting to open a SecureChannel
2020-09-09 12:37.08 debug/channel Requesting to open a SecureChannel
2020-09-09 12:37.08 debug/channel OPN message sent
2020-09-09 12:37.08 debug/client Client Status: ChannelState: OPNSent, SessionState: Closed, ConnectStatus: Good
2020-09-09 12:37.08 trace/client Client connect iterate
2020-09-09 12:37.08 trace/channel Connection 1308 | SecureChannel 0 | Verifying chunk signature
2020-09-09 12:37.08 trace/channel Connection 1308 | SecureChannel 383 | Generating new local keys
2020-09-09 12:37.08 info/channel Connection 1308 | SecureChannel 383 | Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
2020-09-09 12:37.08 info/client Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
2020-09-09 12:37.08 debug/channel Connection 1308 | SecureChannel 383 | Sending request with RequestId 2 of type RegisterServer2Request
2020-09-09 12:37.08 error/channel Connection 1308 | SecureChannel 383 | Received an ERR response with StatusCode BadServiceUnsupported and the following reason: 
2020-09-09 12:37.08 warn/channel Connection 1308 | SecureChannel 383 | Receiving the response failed with StatusCode BadConnectionClosed
2020-09-09 12:37.08 info/client Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadServiceUnsupported
2020-09-09 12:37.08 error/client RegisterServer/RegisterServer2 failed with statuscode BadConnectionClosed
```

Log OPC Foundation LDS:

```
Wed Sep 09 12:37:08 [15620]: ualds_endpoint_callback called: Event=SecureChannelOpened, SecureChanneldId=0x0000017F, uStatus=0x00000000
Wed Sep 09 12:37:08 [15620]: ualds_endpoint_callback: SecureChannel 0x0000017F opened with http://opcfoundation.org/UA/SecurityPolicy#None in mode 1 status 0x00000000!
Wed Sep 09 12:37:08 [15620]: [uastack] OpcUa_SecureListener_ProcessRequest: Closing channel due error 0x800B0000!
Wed Sep 09 12:37:08 [15620]: ualds_endpoint_callback called: Event=SecureChannelOpened, SecureChanneldId=0x00000180, uStatus=0x00000000
Wed Sep 09 12:37:08 [15620]: ualds_endpoint_callback: SecureChannel 0x00000180 opened with http://opcfoundation.org/UA/SecurityPolicy#None in mode 1 status 0x00000000!
Wed Sep 09 12:37:08 [15620]: [uastack] OpcUa_SecureListener_ProcessRequest: Closing channel due error 0x800B0000!
Wed Sep 09 12:37:08 [15620]: ualds_endpoint_callback called: Event=SecureChannelOpened, SecureChanneldId=0x00000181, uStatus=0x00000000
Wed Sep 09 12:37:08 [15620]: ualds_endpoint_callback: SecureChannel 0x00000181 opened with http://opcfoundation.org/UA/SecurityPolicy#None in mode 1 status 0x00000000!
Wed Sep 09 12:37:08 [15620]: [uastack] OpcUa_SecureListener_ProcessRequest: Closing channel due error 0x800B0000!
Wed Sep 09 12:37:08 [15620]: ualds_endpoint_callback called: Event=SecureChannelOpened, SecureChanneldId=0x00000182, uStatus=0x00000000
[... 79 lines cut ...]
```
Log OPC Foundation LDS

```
Wed Sep 09 12:46:42 [15620]: ualds_endpoint_callback called: Event=SecureChannelOpened, SecureChanneldId=0x0000019F, uStatus=0x00000000
Wed Sep 09 12:46:42 [15620]: ualds_endpoint_callback: SecureChannel 0x0000019F opened with http://opcfoundation.org/UA/SecurityPolicy#None in mode 1 status 0x00000000!
Wed Sep 09 12:46:42 [15620]: [uastack] OpcUa_SecureListener_ProcessRequest: Closing channel due error 0x800B0000!
Wed Sep 09 12:46:42 [15620]: ualds_endpoint_callback called: Event=SecureChannelOpened, SecureChanneldId=0x000001A0, uStatus=0x00000000
Wed Sep 09 12:46:42 [15620]: ualds_endpoint_callback: SecureChannel 0x000001A0 opened with http://opcfoundation.org/UA/SecurityPolicy#None in mode 1 status 0x00000000!
Wed Sep 09 12:46:42 [15620]: [uastack] OpcUa_SecureListener_ProcessRequest: Closing channel due error 0x800B0000!
```

I suppose the problem could be with the encryption and signing, but my server(using the same certificate) correct communicate with clients (OPC UA Expert) by Basic128Rsa15 and all other. I would be grateful for every tip. 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1.1
 - [x] Other OPC UA SDKs used (client or server): OPC UA Foundation LDS  v1.04.402.447
 - [x] Operating system: Windows10
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump 
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-09-10:
@Pro Have you seen that before?

mateusz-talma, 2020-09-10:
I found solution.

Before add register callback I assign suitable endpoints and this work fine.
Example code, when work fine with OPC Foundation LDS:

```cpp
m_client = UA_Client_new();
  m_clientConfig = UA_Client_getConfig(m_client);
  UA_ClientConfig_setDefaultEncryption(m_clientConfig, m_certificate, m_privateKey, nullptr, 0, nullptr, 0);
  SetClientInformation(m_clientConfig);
  m_clientConfig->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
 // m_clientConfig->securityPolicyUri = UA_STRING("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15");

  size_t endpointDescriptionsSize = 0;
  UA_EndpointDescription* endpointDescriptions;

  auto statuscode = UA_Client_getEndpoints(m_client, "opc.tcp://localhost:4840", &endpointDescriptionsSize, &endpointDescriptions);

  if (statuscode == UA_STATUSCODE_GOOD)
  {
    for (int i = 0; i < endpointDescriptionsSize; i++)
    {
      if (endpointDescriptions[i].securityMode == UA_MESSAGESECURITYMODE_SIGNANDENCRYPT)
      {
        UA_EndpointDescription_copy(&endpointDescriptions[i], &(m_clientConfig->endpoint));
        break;
      }
    }
    UA_Array_delete(endpointDescriptions, endpointDescriptionsSize, &UA_TYPES[UA_TYPES_ENDPOINTDESCRIPTION]);
  }

  auto status = UA_Server_addPeriodicServerRegisterCallback(
    m_serverUA, m_client, "opc.tcp://localhost:4840", 10 * 60 * 1000, 500, nullptr);
```

Maybe the documentation needs to be expanded or prepare additional example for server register operation?

teccheck, 2025-05-21:
Since 1.4 we certified against full server profile 2017. This also includes LDS tests. We also use the official LDS. We think all issues are solved. Feel free to reopen if needed