open62541/open62541 #3881: Problems passing CTT on Windows Machine

Wooodys, 2020-09-10:
## Description
I'm trying to pass the CTT with my application using open62541 on Release 1.1. But I have problems passing some test, even when I just use the server_ctt.c example. Is there some routine that I need to start the tests. 

The following tests failed:
* Base Information -> Base Info Core Structure -> 001.js -> ProgramStateMachineType -> ProgramDiagnostics ==> [NOT FOUND] REQUIRED (That's strange, because the node is actually on the server when I browse it with UA Expert)
* Monitored Item Services -> Monitor QueueSize_1 ->005.js -> Expected to receive a dataChange notification (test passes around 7 out of 10 times)

## Background Information / Reproduction Steps 
I'm using a Windows 10 Machine and tried with both CTT Versions 1.3.341.396 and 1.4.9.396. I use the xml settings open62541_certify.ctt.xml and selection open62541_certify.selection.xml provided by the open62541_ctt branch. I start the server_ctt example with all certificates provided by the ctt and some selfsigned certificates.

Used CMake options:

```bash
UA_ARCHITECTURE:STRING=win32
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=ON
UA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=OFF
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=ON
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_PARSING:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_LOGLEVEL:STRING=200
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
```

## Checklist

Please provide the following information:

 * open62541 Version (release number or git tag): v1.1
 - [ ] Other OPC UA SDKs used (client or server): 
 * Operating system:Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-09-10:
Certification for 1.0 was done with the CTT 1.03.341.389.

> BadTooManySessions 

Some issues might be due to the configuration of the CTT case.
There are settings for max sessions. Both in the CTT and in the open62541 SDK.

The other issues we have to look into.

Wooodys, 2020-09-10:
Thanks for the feedback. 

I was able to resolve some issues by editing the CTT settings to match the server ones. But not the first one. I also detect some problems with failing tests while repeating them. 

I'm not sure if I got the settings right. For example some CTT settings seems to be not adjustable by the open62541 SDK:
* Session DefaultTimeoutHint
* Secure Channel NetworkTimeout (is it maxSecurityTokenLifetime?)

Wooodys, 2020-09-15:
I finished testing my application and I just wann mention some small things, that I needed to change to pass the tests.

> Base Information -> Base Info Core Structure -> 001.js -> ProgramStateMachineType -> ProgramDiagnostics ==> [NOT FOUND] REQUIRED 

Under CTT 1.3... it's just a warning.

> Session Services -> Session Base -> 002.js and 012.js -> Bad: BadTooManySessions (if I ran test 012.js solo it finish successfully. But if 002.js is run before 012.js, it fails around 7 out of 10 times.

The server cleanup is in a repeated callback that is by default executed every 10 seconds. Test 002.js checks if the session has timed out and continues. The session is not cleared at this point. When you have bad luck, it's still not cleared when 012.js starts to open the maximum amount of sessions. To solve this, I just set the repeated callback to 1 second. 

> Monitored Item Services -> Monitor QueueSize_1 ->005.js -> Expected to receive a dataChange notification (test passes around 7 out of 10 times)

Not sure what makes this one fail. It does have something to do with the Datatype of the nodes. Just happens with Boolean. 

> Security -> Security Certificate Validation -> 010.js -> Certificate with invalid signiture is still accepted by the server.

Not sure if it is a bug or a feature of mbedtls. Under mbedtls version 2.7... it works just fine. Everything newer doesn't detect the invalid signiture.

closed 2025-05-21 by teccheck