open62541/open62541 #3882: Problems with debugging open62541

MichaelKlin, 2020-09-10:
Hi,
open62541 Version 1.1.2(not the single-file distribution), Windows 10 Pro
Using Cmake and Visual Studio 14 2015, I can successfully generate a build for open62541. I can open the project in Visual Studio and it will build successfully. I can also start exe. of Examples However, when I attempt to debug it, I get the following error: Unable to start program 'C:\open62541\build\x64\Debug\ALL_BUILD'. Access is denied. I cannot figure out what is causing this to occur.
I use cmake UA_BUILD_EXAMPLES and UA_ENABLE_SUBSCRIPTIONS

cmake -DUA_BUILD_EXAMPLES -DUA_ENABLE_SUBSCRIPTIONS

How can i debug my code?
Thanks in advance.

## Checklist

Please provide the following information:

 - [+] open62541 Version (release number or git tag):
 - [+] Other OPC UA SDKs used (client or server): 
 - [+] Operating system:
 - [-] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [-] Wireshark network dump attached
 - [-] Self-contained code example attached
 - [+] Critical issue

viktor-morin, 2020-09-10:
Access is denied.
Are you running Visual Studio with admin privilege?

MichaelKlin, 2020-09-11:
@viktor-morin I try it with admin privelege. But i get the same error. Programm myprojeckt\open62541\server\Debug\ALL_BUILD can not start, System can not find this File
[image]

viktor-morin, 2020-09-11:
@MichaelKlin 
It dosn't even look like a file? ALL_BUILD
ALL_BUILD is used to config the CMake files etc. 
[https://stackoverflow.com/questions/27175754/what-are-all-build-and-zero-check-and-do-i-need-them](https://stackoverflow.com/questions/27175754/what-are-all-build-and-zero-check-and-do-i-need-them)

ALL_BUILD is not a project you can run.

MichaelKlin, 2020-09-11:
@viktor-morin yes, i solved my problem. Thanks you

closed 2020-09-11 by MichaelKlin