open62541/open62541 #3886: Publish/Subscribe does not work?

MichaelKlin, 2020-09-14:
## Description
I install open62541 -actual master(v1.1.2) and use Windows10 pro.
I start tutorial_pubsub_publish and tutorial_pubsub_subscribe and now i recognize, that i do not get the messages from publish. See Pics. I am not sure, is it right or something wrong?

Can you maybe please say, what is difference between tutorial_pubsub_subscribe/publish and server_pubsub_publish_rt_level/subscribe or can i use them together?

In tutorial_pubsub_subscribe i see, that there is used hardcoded DataSetMetadate, Can i change DataSetMetadate dynamiclly. For Example: I publish every times different datatype and change dataSetmetadate, 

 And last samll question, is it possible, that client published and server subscribed?

[image]
[image]

## Background Information / Reproduction Steps
 start tutorial_pubsub_publish and tutorial_pubsub_subscribe

Used CMake options:

Commandline options:
-DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL="1" -DUA_ENABLE_PUBSUB:BOOL="1" -DUA_NAMESPACE_ZERO:STRING="FULL" -DUA_BUILD_EXAMPLES:BOOL="1" -DUA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL="1" 

Cache file:
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=1
UA_ENABLE_PUBSUB:BOOL=1
UA_NAMESPACE_ZERO:STRING=FULL
UA_BUILD_EXAMPLES:BOOL=1
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=1

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [+] open62541 Version (release number or git tag):
 - [+] Other OPC UA SDKs used (client or server): 
 - [+] Operating system:
 - [+] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [+] Critical issue

jackybek, 2021-02-11:
There are elements in the structure under the pubsub section (open62541.h) that needs to be initialised before the function will work.

NoelGraf, 2021-04-22:
@EvgKlin Have you tried it again?

NoelGraf, 2021-04-29:
I would close the issue for now. Please reopen if required!

closed 2021-04-29 by NoelGraf