open62541/open62541 #3891: Duplicate files in fuzzing corpus

mlgiraud, 2020-09-16:
## Description
The current fuzzing corpus contains duplicate files (the content is an exact duplicate).
For most fuzzers this is not ideal. @Pro I think you did some of the fuzzing stuff, right?
Any reason for the duplicates, or is this just an oversight?

I would propose to remove all duplicates listed here:

<details>
<summary>Duplicates list</summary>
```
./server_historical_data_00028_msg_CreateSessionRequest.bin
./server_historical_data_00021_msg_CreateSessionRequest.bin

./discovery_00104_opn.bin
./discovery_00108_opn.bin

./server_historical_data_00027_msg_GetEndpointsRequest.bin
./server_historical_data_00020_msg_GetEndpointsRequest.bin

./server_historical_data_00083_clo.bin
./server_historical_data_00062_clo.bin
./server_historical_data_00076_clo.bin
./server_historical_data_00069_clo.bin
./server_historical_data_00048_clo.bin
./server_historical_data_00055_clo.bin

./discovery_00010_opn.bin
./discovery_00002_opn.bin
./discovery_00006_opn.bin
./client_async_00109_opn.bin

./client_00072_opn.bin
./client_00066_opn.bin

./monitoreditem_filter_00069_opn.bin
./monitoreditem_filter_00059_opn.bin

./monitoreditem_filter_00073_msg_CreateSubscriptionRequest.bin
./monitoreditem_filter_00063_msg_CreateSubscriptionRequest.bin

./encryption_basic256sha256_00004_clo.bin
./encryption_basic128rsa15_00004_clo.bin
./client_00018_clo.bin

./client_00013_msg_CloseSessionRequest.bin
./client_00006_msg_CloseSessionRequest.bin

./client_historical_data_00029_msg_CloseSessionRequest.bin
./client_historical_data_00059_msg_CloseSessionRequest.bin

./discovery_00112_opn.bin
./discovery_00116_opn.bin
./discovery_00120_opn.bin

./client_highlevel_00053_msg_CloseSessionRequest.bin
./client_highlevel_00015_msg_CloseSessionRequest.bin

./accesscontrol_00026_clo.bin
./accesscontrol_00020_clo.bin

./server_callbacks_00021_msg_ActivateSessionRequest.bin
./server_callbacks_00029_msg_ActivateSessionRequest.bin
./monitoreditem_filter_00005_msg_ActivateSessionRequest.bin
./server_callbacks_00005_msg_ActivateSessionRequest.bin
./server_callbacks_00013_msg_ActivateSessionRequest.bin

./server_historical_data_00019_opn.bin
./server_historical_data_00026_opn.bin

./client_historical_data_00060_clo.bin
[... 374 lines cut ...]
./client_highlevel_00046_hel.bin
./client_async_00117_hel.bin
./client_async_00126_hel.bin
./server_historical_data_00063_hel.bin
./discovery_00035_hel.bin
./monitoreditem_filter_00031_hel.bin
./discovery_00009_hel.bin
./discovery_00119_hel.bin
./monitoreditem_filter_00068_hel.bin
./server_callbacks_00009_hel.bin
./client_async_00108_hel.bin
./client_00019_hel.bin
./accesscontrol_00008_hel.bin
./client_securechannel_00025_hel.bin
./discovery_00031_hel.bin
./encryption_basic128rsa15_00001_hel.bin
./monitoreditem_filter_00145_hel.bin
./client_securechannel_00020_hel.bin

./client_historical_data_00070_msg_HistoryReadRequest.bin
./client_historical_data_00182_msg_HistoryReadRequest.bin
./client_historical_data_00194_msg_HistoryReadRequest.bin
./client_historical_data_00082_msg_HistoryReadRequest.bin
```
</details>

Pro, 2020-09-21:
Yes, I did most of the fuzzing setup.

This corpus is generated by the script mentioned here:
https://github.com/open62541/open62541/tree/master/tests/fuzz#update-the-corpus
This also results in multiple duplicates.

This corpus is then automatically merged with the corresponding clang fuzzer script to remove duplicates, and to integrate new files into the existing fuzzer corpus:
https://github.com/open62541/open62541/blob/master/tests/fuzz/generate_corpus.sh#L120

Therefore it shouldn't matter if there are duplicates.
Feel free to suggest improvements to this approach :)

NoelGraf, 2021-04-22:
What is the status of the issue? Can the issue be closed?

NoelGraf, 2021-04-29:
I would close the issue for now. Please reopen if anyone has improvements to suggest this approach or has problems regarding this approach.

closed 2021-04-29 by NoelGraf