open62541/open62541 #3893: UA_Client_Service_registerNodes and UA_Client_Service_read from different clients

luibass92, 2020-09-16:
## Description

Hello everyone,

I found out that a correct way to have more fast responses from the server is to perform a `UA_Client_Service_registerNodes` before the `UA_Client_Service_read`.

I managed to use correctly both APIs but I was wondering if they MUST be used by the same client or if they can be used from different clients obtaining the same result.

e.g.
Let's suppose I have a 1st Client which performs just one time `UA_Client_Service_registerNodes`  on a given set of nodes and, later, a 2nd Client performs several `UA_Client_Service_read` on the same given set of nodes.
In this case is the 2nd Client exploiting the 1st Client call to `UA_Client_Service_registerNodes` or not ?

## Background Information / Reproduction Steps

Following an example code snippet in which **CLIENT_1** performs `UA_Client_Service_registerNodes` and **CLIENT_2** performs `UA_Client_Service_read` on the previously registered node :

```
UA_RegisterNodesRequest req;
UA_RegisterNodesRequest_init(&req);
req.nodesToRegister = UA_NodeId_new();
req.nodesToRegister[0] = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_ CURRENTTIME);
req.nodesToRegisterSize = 1;
UA_RegisterNodesResponse res = UA_Client_Service_registerNodes(CLIENT_1, req);

UA_ReadRequest rreq;
UA_ReadRequest_init(&rreq);
rreq.nodesToRead = UA_NodeId_new();
rreq.nodesToRead[0] = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_ CURRENTTIME);
req.nodesToReadSize = 1;
UA_ReadResponse rres = UA_Client_Service_read(CLIENT_2, rreq);

```

Is CLIENT_2 actually reading a registered node? Is the reading actually faster?

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.1.1-7-g5ad7c7fc
 - [ ] Operating system: Windows 10 / Ubuntu 18.04
 - [ ] Self-contained code example attached

ccvca, 2020-09-17:
According to the specification, the requests must de done using the same session:
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.8.5/

luibass92, 2020-09-17:
> According to the specification, the requests must de done using the same session:
> https://reference.opcfoundation.org/v104/Core/docs/Part4/5.8.5/

Thanks @ccvca .
Do you know if, in case of unplanned client disconnections, the server automatically frees up the resources allocated for a previous nodes registration of that client?

ccvca, 2020-09-17:
This depends on the server implementation.

closed 2020-09-17 by luibass92