open62541/open62541 #3894: Datachange Source (Client vs. Server)

ArriKing, 2020-09-17:
## Description
I'm looking for a way to differentiate between data changes that were made by a client connection and those which were made by the server itself.  I'm looking to have my server update its nodes without triggering a bunch of datachange events, but at the same time know that a client is writting new data.

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_ARCHITECTURE_POSIX
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag):1.1.2
 - [X] Other OPC UA SDKs used (client or server): UaExpert
 - [X] Operating system:Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

juangburgos, 2020-10-08:
You can subscribe an `afterWrite` callback and distinguish who wrote the variable's value, for example by checking the session nodeId.

* <https://open62541.org/doc/current/tutorial_server_datasource.html#variable-value-callback>

NoelGraf, 2021-04-22:
What is the status of the issue? Can the issue be closed?

juangburgos, 2021-04-22:
I think so, yes

closed 2021-04-29 by NoelGraf