open62541/open62541 #3895: UA_Client_Service_browse() doesn't get all hierarchical references

skubidu1324, 2020-09-17:
## Description
I write a client that should browse a server. I used the  function `UA_Client_Service_browse` but doesn't get all references. 

## Background Information / Reproduction Steps
I followed the steps from the examples with the following parameters:

```
UA_BrowseRequest bReq;
UA_BrowseRequest_init(&bReq);
bReq.requestedMaxReferencesPerNode = 0;
bReq.nodesToBrowse = UA_BrowseDescription_new();
bReq.nodesToBrowseSize = 1;
bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL;
bReq.nodesToBrowse[0].includeSubtypes = UA_TRUE;
bReq.nodesToBrowse[0].referenceTypeId = UA_NODEID_NUMERIC(0, UA_NS0ID_HIERARCHICALREFERENCES);
```

[image]
I've browsed that node and got only one reference. (the one with `HasProperty`) The other two with 'Organizes' are missing although it is a subtype of the hierachical references.

[image]

If I change it to

```
bReq.nodesToBrowse[0].referenceTypeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
```

I got the two missing references. Is that a bug or do I miss something? Thanks in advance!

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 2d2b98b
 - [x] Other OPC UA SDKs used (client or server): server Siemens PLC , client open62541
 - [x] Operating system: windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-09-22:
Hmm. Doesn't sound good.

I added a test for the referencetypes subtyping.
https://github.com/open62541/open62541/pull/3905

That doesn't show any issue. Will investigate further.

jpfr, 2020-09-27:
I added a test specifically for the case you reported.
https://github.com/open62541/open62541/pull/3916/commits/085ead4c178eafe2948429bc3ae99079359b6786
All works well.

Only now did I see that you report *Siemens PLC* for the server-side SDK.
I believe the problem is the server-side, not the client side.

We do not do any pre- or post-processing with the Browse request or response.

Please take this up with the server SDK vendor.

closed 2020-09-27 by jpfr

Smitter, 2020-10-09:
I have the same issue. That's what i've found so far...
1. UA_BrowseResponse_clear(&bResp); // Leads to crash if bResp is empty. Trace shows unlogical calls
2. UA_Client_Service_browse(...) will actually send telegram "DeleteSubscriptionsRequest", this i can see with Wireshark and also this i see while tracing into sendSymmetricServiceRequest(), this is what is written in requestType.typeName.

So, it looks like UA_TYPES[UA_TYPES_BROWSEREQUEST] contains incorrect callbacks.

skubidu1324, 2020-10-30:
I have the solution for that problem. It has nothing to do with the references but with the `continuationPoint`. I thought that if  `bReq.requestedMaxReferencesPerNode  == bResp->results[0].referencesSize ` then I have to make a `BrowseNext `. That is not the case. You have to it if the `bResp->results[0].continuationPoint` is filled.

I came to this solution by inspecting the behaviour of UAExpert with wireshark...