open62541/open62541 #3904: Server Variable can not be changed

MichaelKlin, 2020-09-22:
## Description
Hello,
 I was able to read  a variable from server on the client, I am simply just increment the variable on the client by one and am writing it onto the variable using UA_Client_writeValueAttribute or UA_Client_Service_write. But If i read variable from server, Variable is not changed. Server has old Variable-values.  
Thanks!

## Background Information / Reproduction Steps

 UA_WriteResponse wResp = UA_Client_Service_write(client, wReq);
    if(wResp.responseHeader.serviceResult == UA_STATUSCODE_GOOD)
        printf("the new value is: %i\n", value);
or
   UA_Client_writeValueAttribute(client, UA_NODEID_STRING(1, "the.answer"), myVariant);

    status =UA_Client_readValueAttribute(client, UA_NODEID_STRING(1, "the.answer"), &val);
    if(status == UA_STATUSCODE_GOOD &&
       UA_Variant_hasScalarType(&val, &UA_TYPES[UA_TYPES_INT32])) {
        value = *(UA_Int32 *)val.data;
        printf("the server value is: %i\n", value);   }

Logs on client

Reading the value of node (1, "the.answer"):
the old server value is: 42
Writing a value of node (1, "the.answer"):
the new value is: 43

the server value is: 42

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): V1.1.2
 - [x] Operating system: Windows
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached

jpfr, 2020-09-27:
What is the return StatusCode of the write operation?
It could be you are sending the wrong datatype. Or that the variable is not writeable.

MichaelKlin, 2020-11-09:
I close it

closed 2020-11-09 by MichaelKlin

Youngmingtao, 2022-10-12:
> 

I'm facing the same problem, could I ask how did you solve the problem, plenty thanks.

Elendil211, 2023-01-04:
I'm facing the same problem, too. Were you able to fix it?