open62541/open62541 #3906: small memory leak on processServiceResponse when I use the UA_Client_Service_read method

I-Yoo, 2020-09-23:
open62541 Version release 1.1, windows, TCP

jpfr, 2020-09-24:
Use Dr. Memory to produce some indication on the origin of the leak.

I-Yoo, 2020-09-25:
I debugged in Visual Studio 2017 using a memory snapshot for heap analysis and found that a memory leak occurred during the following call
[
[image]
](url)

I-Yoo, 2020-09-25:
I debugged in Visual Studio 2017 using a memory snapshot for heap analysis and found a memory leak in the following call structure. There is a linear relationship between the count difference and the number of nodes I request to read.
[image]

I-Yoo, 2020-09-25:
Is the problem in the diagram in Array_decodeBinary and UA_new?

jpfr, 2020-09-26:
Do you clean up the value returned from UA_Client_Service_read?

closed 2020-09-27 by I-Yoo