open62541/open62541 #3910: writeDataTypeAttribute do not wort

EugenM-sie, 2020-09-24:
## Description

On Client side i want to change the data type. I always get this mistake statuscode 2147614720

## Background Information / Reproduction Steps

I create Object and Variable node on client side with UA_Client_addObjectNode or UA_Client_addVariableNode. My Nodes Variable are Int32 and Uint32.

After that, I write that:

       UA_NodeId newDataType;   //or             UA_NodeId newDataType=UA_NODEID_STRING(1, "testid")
       newDataType.identifierType = UA_NODEIDTYPE_STRING;
      
       status = UA_Client_writeDataTypeAttribute(client, UA_NODEID_NUMERIC(1, 50510),
                                                  &newDataType);
        if(status == UA_STATUSCODE_GOOD) {
            printf("Changed datatype");
        }

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): Master V1.1.2
 - [x] Operating system: Windows
 - [x] Self-contained code example attached

jpfr, 2020-09-26:
Can you provide the log output?
Probably you still have a value attribute in the node that is not a string.

EugenM-sie, 2020-09-29:
@jpfr  Hallo, Thank you, i fixed it by changing the value attribute. Can you maybe help me? I try to send a pumpobjeckt from tutorial by publisch/subscribe connection. In Wireshark can i see messages to target 224.0.0.22:4840, but i am not sure, if i subscribe right object and how can i objeckt represent. Thank you

teccheck, 2025-05-21:
We have a test for that in `check_client_highlevel`, so it should not be an issue anymore