open62541/open62541 #3921: Lossless high speed MonitoredItems notifications

jugurthab, 2020-09-29:
Hi,
I'm setting up the following configuration for my project :

Telegraf => open62541 server => OPC UA client.

Telegraf updates a set of variables in open62541 server at a very high speed rate. I'd like to get all these changes on OPC UA client (without loosing any of them).

Note : The client should receive all the changes without changing it's source code.

I have tried MonitoredItems : https://open62541.org/doc/current/tutorial_server_monitoreditems.html (I used the exact configuration provided in the example) but I find it slow when I look at OPC UA client logs.

Thank's.

jpfr, 2020-09-29:
OPC UA has two kinds of MonitoredItems:
- DataChanges
- Events

DataChange-MonitoredItems rely on a sampling interval.
You want that as least twice as fast as the incoming data frequency in order to lose nothing.

However, Event-MonitoredItems, that don't have a sampling interval, are a much better fit.

In addition, make sure you have the queues configured right so that a temporary loss of connection does not lead to data loss.

jugurthab, 2020-09-30:
Thanks for your response @jpfr , that really helped me to choose.
I have a last question :
I followed this tutorial : https://open62541.org/doc/current/tutorial_server_events.html but I didn't see anything on UaExpert unless I call the function that triggers the event explicitly (and yet, it only prints "receive event").

In my situation, I have 1 object that contains many variables and I'm interested in these variables:
- should I attach the event to every variable or just their parent object (which has eventNotifier enabled)?
- How to trigger the event everytime variables value changes (without explicitly calling the function)?

Thank's a lot.

basyskom-meerkoetter, 2020-10-07:
wouldn't this also be a nice fit for pub/sub?

teccheck, 2025-05-21:
There's been no recent activity and it seems to be userland issue. If you're still working on it, check our docs and examples. They have been updated a lot since then. Feel free to reopen this issue

closed 2025-05-21 by teccheck