open62541/open62541 #3924: Unable to enable the UA_ENABLE_ENCRYPTION ON and Encryption of mbedtls

Yashvanthb13, 2020-09-30:
Hi all

I'm unable to enable the encryption on   i used the Cmake gui to enable, I followed below steps for code generation

1. git clone of open62541 
2. cd open65241, mkdir build and cd  build
3. cmake gui with location build folder and opened open62541.sln and rebuild the projects

but to enable the encryption, but when i click on the configure button i got this errors.

i made UA_ENABLE_ENCRYPTION enable in config.h file also.

CMake Error at C:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find MbedTLS (missing: MBEDTLS_INCLUDE_DIRS MBEDTLS_LIBRARY
  MBEDX509_LIBRARY MBEDCRYPTO_LIBRARY)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  tools/cmake/FindMbedTLS.cmake:35 (find_package_handle_standard_args)
  CMakeLists.txt:487 (find_package)

can anyone provide the steps for enable encryption with openssl or mbedtls

juangburgos, 2020-09-30:
I was trying to update [QUaServer](https://github.com/juangburgos/QUaServer) to the latest master and got the same nice surprise. Could not find any info on the docs either. I made it work by defining those extra CMake flags:

```
-DUA_ENABLE_ENCRYPTION_MBEDTLS=ON -DMBEDTLS_INCLUDE_DIRS=${MBEDTLS_PATH}/include -DMBEDTLS_FOLDER_LIBRARY=${MBEDTLS_PATH}/build/library/Debug
```

If you are building for release, then you might wanna change `MBEDTLS_FOLDER_LIBRARY` for the release version.

juangburgos, 2020-09-30:
To see how to generate certificates, maybe [this link](https://github.com/juangburgos/QUaServer#create-certificates) helps you.

Yashvanthb13, 2020-09-30:
Thanks for the replay @juangburgos 
can you also provide the steps for building the VS code with mbedtls and encryption enabled

juangburgos, 2020-09-30:
Not much time to give you explicit steps, but you can infer from my qmake scripts:

* <https://github.com/juangburgos/QUaServer/blob/462b2f2d3791ffe9bb0a68b3ffd57cef1930be35/src/amalgamation/open62541encryption.pri#L80>

* <https://github.com/juangburgos/QUaServer/blob/462b2f2d3791ffe9bb0a68b3ffd57cef1930be35/src/amalgamation/open62541amalgamation.pri#L126>

jpfr, 2020-10-01:
> how to generate the certificate files for connection with server in encryption format

We have a script for that in /tools/certs.

jpfr, 2020-10-01:
The MbedTLS library is detected via the script in /tools/cmake/FindMbedTLS.cmake.

Yashvanthb13, 2020-10-02:
> The MbedTLS library is detected via the script in /tools/cmake/FindMbedTLS.cmak

should i download the mbedtls and add that path or will it auto detect because, in Cmake it not able to dectect when try to generate the code of VS

juangburgos, 2020-10-05:
The "detection" mechanisms are not very robust, and of course they won't work in my case, when I build mbedtls from source on windows. I suggest you define the `MBEDTLS_INCLUDE_DIRS` and `MBEDTLS_FOLDER_LIBRARY` manually as I did to make it work.

Yashvanthb13, 2020-10-06:
I add the Mbedtls paths of library manually in cmake, after generating code i rebuild the ALL_BUILD after the i got some Linker errors related to mbedtls  

likesecuritypolicy_mbedtls_common.obj : error LNK2019: unresolved external symbol _mbedtls_md_get_size referenced in function _mbedtls_generateKey

13>securitypolicy_mbedtls_common.obj : error LNK2019: unresolved external symbol _mbedtls_md_hmac_starts referenced in function _mbedtls_hmac
13>securitypolicy_mbedtls_common.obj : error LNK2019: unresolved external symbol _mbedtls_md_hmac_update referenced in function _mbedtls_hmac
13>securitypolicy_mbedtls_common.obj : error LNK2019: unresolved external symbol _mbedtls_md_hmac_finish referenced in function _mbedtls_hmac
13>ua_securitypolicy_aes128sha256rsaoaep.obj : error LNK2001: unresolved external symbol _mbedtls_rsa_set_padding
13>securitypolicy_mbedtls_common.obj : error LNK2001: unresolved external symbol _mbedtls_rsa_set_padding
13>ua_securitypolicy_basic128rsa15.obj : error LNK2001: unresolved external symbol _mbedtls_rsa_set_padding
13>ua_securitypolicy_basic256.obj : error LNK2001: unresolved external symbol _mbedtls_rsa_set_padding
13>ua_securitypolicy_basic256sha256.obj : error LNK2001: unresolved external symbol _mbedtls_rsa_set_padding
13>securitypolicy_mbedtls_common.obj : error LNK2019: unresolved external symbol _mbedtls_rsa_rsaes_oaep_encrypt referenced in function _mbedtls_encrypt_rsaOaep
13>securitypolicy_mbedtls_common.obj : error LNK2019: unresolved external symbol _mbedtls_rsa_rsaes_oaep_decrypt referenced in function _mbedtls_decrypt_rsaOaep
13>securitypolicy_mbedtls_common.obj : error LNK2019: unresolved external symbol _mbedtls_pk_verify referenced in function _mbedtls_verifySig_sha1
13>ua_securitypolicy_basic256sha256.obj : error LNK2001: unresolved external symbol _mbedtls_pk_verify
13>ua_securitypolicy_aes128sha256rsaoaep.obj : error LNK2001: unresolved external symbol _mbedtls_pk_verify
13>securitypolicy_mbedtls_common.obj : error LNK2019: unresolved external symbol _mbedtls_pk_sign referenced in function _mbedtls_sign_sha1
13>ua_securitypolicy_basic256sha256.obj : error LNK2001: unresolved external symbol _mbedtls_pk_sign
13>ua_securitypolicy_aes128sha256rsaoaep.obj : error LNK2001: unresolved external symbol _mbedtls_pk_sign
13>ua_securitypolicy_aes128sha256rsaoaep.obj : error LNK2001: unresolved external symbol _mbedtls_ctr_drbg_random
13>securitypolicy_mbedtls_common.obj : error LNK2001: unresolved external symbol _mbedtls_ctr_drbg_random
13>ua_securitypolicy_basic128rsa15.obj : error LNK2001: unresolved external symbol _mbedtls_ctr_drbg_random
13>ua_securitypolicy_basic256.obj : error LNK2001: unresolved external symbol _mbedtls_ctr_drbg_random
13>ua_securitypolicy_basic256sha256.obj : error LNK2001: unresolved external symbol _mbedtls_ctr_drbg_random
13>securitypolicy_mbedtls_common.obj : error LNK2019: unresolved external symbol _mbedtls_sha1_ret referenced in function _mbedtls_verifySig_sha1
13>ua_securitypolicy_basic128rsa15.obj : error LNK2019: unresolved external symbol _mbedtls_md_info_from_type referenced in function _policyContext_newContext_sp_basic128rsa15
13>ua_securitypolicy_basic256.obj : error LNK2001: unresolved external symbol _mbedtls_md_info_from_type
13>ua_securitypolicy_basic256sha256.obj : error LNK2001: unresolved external symbol _mbedtls_md_info_from_type
13>ua_securitypolicy_aes128sha256rsaoaep.obj : error LNK2001: unresolved external symbol _mbedtls_md_info_from_type
13>ua_securitypolicy_basic128rsa15.obj : error LNK2019: unresolved external symbol _mbedtls_md_init referenced in function _policyContext_newContext_sp_basic128rsa15
13>ua_securitypolicy_basic256.obj : error LNK2001: unresolved external symbol _mbedtls_md_init
13>ua_securitypolicy_basic256sha256.obj : error LNK2001: unresolved external symbol _mbedtls_md_init
13>ua_securitypolicy_aes128sha256rsaoaep.obj : error LNK2001: unresolved external symbol _mbedtls_md_init
13>ua_securitypolicy_basic128rsa15.obj : error LNK2019: unresolved external symbol _mbedtls_md_free referenced in function _clear_sp_basic128rsa15
13>ua_securitypolicy_basic256.obj : error LNK2001: unresolved external symbol _mbedtls_md_free
13>ua_securitypolicy_basic256sha256.obj : error LNK2001: unresolved external symbol _mbedtls_md_free
13>ua_securitypolicy_aes128sha256rsaoaep.obj : error LNK2001: unresolved external symbol _mbedtls_md_free
13>ua_securitypolicy_basic128rsa15.obj : error LNK2019: unresolved external symbol _mbedtls_md_setup referenced in function _policyContext_newContext_sp_basic128rsa15
13>ua_securitypolicy_basic256.obj : error LNK2001: unresolved external symbol _mbedtls_md_setup
13>ua_securitypolicy_basic256sha256.obj : error LNK2001: unresolved external symbol _mbedtls_md_setup
13>ua_securitypolicy_aes128sha256rsaoaep.obj : error LNK2001: unresolved external symbol _mbedtls_md_setup
13>ua_securitypolicy_basic128rsa15.obj : error LNK2019: unresolved external symbol _mbedtls_pk_init referenced in function _updateCertificateAndPrivateKey_sp_basic128rsa15
13>ua_securitypolicy_basic256.obj : error LNK2001: unresolved external symbol _mbedtls_pk_init
13>ua_securitypolicy_basic256sha256.obj : error LNK2001: unresolved external symbol _mbedtls_pk_init
13>ua_securitypolicy_aes128sha256rsaoaep.obj : error LNK2001: unresolved external symbol _mbedtls_pk_init
[... cut ...]
13>ua_pki_mbedtls.obj : error LNK2019: unresolved external symbol _mbedtls_x509_crl_free referenced in function _certificateVerification_clear
13>ua_pki_mbedtls.obj : error LNK2019: unresolved external symbol _mbedtls_x509_crt_verify_info referenced in function _certificateVerification_verify
13>ua_pki_mbedtls.obj : error LNK2019: unresolved external symbol _mbedtls_x509_crt_verify_with_profile referenced in function _certificateVerification_verify

i have added required files and libs file path can anyone provide solution on this

juangburgos, 2020-10-06:
The `unresolved external symbol` means you are not adding the mbedtls library binary correctly. Make sure you have a compatible library (compiled with the same compiler, supported mbedtls version and that the `MBEDTLS_FOLDER_LIBRARY` points correctly to that binary. I use version 2.17.0 of mbedtls, build with visual studio and links correctly with open62541.

Yashvanthb13, 2020-10-06:
before adding MBEDTLS_FOLDER_LIBRARY i complied and added them, i have used mbedtls-2.24.0

juangburgos, 2020-10-06:
I have no more tips to give you. I have compiled latest open62541 master with VisualStudio 2013 and 2017, both for 32 and 64 bits, using mbedtls version 2.17.0 and compiles perfectly. You have to figure out yourself what you are doing wrong. Good luck.

Yashvanthb13, 2020-10-06:
now i now build and connection with server is happing with none and without username and password when i try to connect with user name and password i'm getting this type of errors, encryption is enabled.

without using certs I was trying to connect.

any commands there  for certificate generation which is in tool/certs

    Rejecting UserTokenPolicy 0 (anonymous) in endpoint 0: configuration doesn't match
    Rejecting UserTokenPolicy 1 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
    Rejecting UserTokenPolicy 2 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
    Rejecting endpoint 1: security policy not available
    Rejecting endpoint 2: security policy not available
    Rejecting endpoint 3: security policy not available
    Rejecting endpoint 4: security policy not available
    No suitable UserTokenPolicy found for the possible endpoints
     Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError

where i'm doing wrong

Yashvanthb13, 2020-10-07:
Now i am able to connect with user name i used some sample certs now i need to generate the own certs using /tools/certs/localhost.cnf
can anyone know the steps to generate the certs

Yashvanthb13, 2020-10-07:
Getting Error in connection with pro sys server using certificate, User name and password 
Below is the sample code i used for connection with server

	#include <open62541/client_config_default.h>
	#include <open62541/client_highlevel.h>
	#include <open62541/client_subscriptions.h>
	#include <open62541/plugin/log_stdout.h>

	#include "common.h"

	int main(int argc, char *argv[])
	{
		UA_String securityPolicyUri = UA_STRING("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");

		UA_MessageSecurityMode securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT; /* allow everything */

		char *serverurl = "opc.tcp://192.168.1.102:4840";
		
		char *username = "test";
		
		char *password = "test";

	#ifdef UA_ENABLE_ENCRYPTION
		char *certfile = "..\\certs\\server_cert.der";
		char *keyfile = "..\\certs\\server_key.der";
		 /* Create the server and set its config */
		UA_Client *client = UA_Client_new();
		UA_ClientConfig *cc = UA_Client_getConfig(client);

	#ifdef UA_ENABLE_ENCRYPTION
		if(certfile) {
			UA_ByteString certificate = loadFile(certfile);
			UA_ByteString privateKey  = loadFile(keyfile);
			UA_ClientConfig_setDefaultEncryption(cc, certificate, privateKey,
												 NULL, 0, NULL, 0);
			UA_ByteString_clear(&certificate);
			UA_ByteString_clear(&privateKey);
		} else {
			UA_ClientConfig_setDefault(cc);
		}
	#else
		UA_ClientConfig_setDefault(cc);
	#endif

		cc->securityMode = securityMode;
		cc->securityPolicyUri = securityPolicyUri;

		/* Connect to the server */
		UA_StatusCode retval = UA_STATUSCODE_GOOD;
		if(username) 
		{
			retval = UA_Client_connectUsername(client, serverurl, username, password);
		}
		else 
		{

			retval = UA_Client_connect(client, serverurl);
		}

		if(retval != UA_STATUSCODE_GOOD) 
		{
			UA_Client_delete(client);
			return EXIT_FAILURE;
		} 
		else 
		{
			printf("Connection establishment successful \r\n");

			UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Connected!");

[... 21 lines cut ...]
"minkernel crts ucrt src appcrt heap debug_heap.cpp line 904"

"server security policy #base256sha256 , mode is sign, sign&encrypt"

[image]

	[2020-10-07 12:45:54.069 (UTC+0550)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
	[2020-10-07 12:45:54.075 (UTC+0550)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
	[2020-10-07 12:45:54.076 (UTC+0550)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
	[2020-10-07 12:45:54.076 (UTC+0550)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
	[2020-10-07 12:45:54.076 (UTC+0550)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
	[2020-10-07 12:45:54.077 (UTC+0550)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
	[2020-10-07 12:45:55.985 (UTC+0550)] info/channel       Connection 268 | SecureChannel 5 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
	[2020-10-07 12:45:55.985 (UTC+0550)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
	[2020-10-07 12:45:55.999 (UTC+0550)] info/client        Rejecting UserTokenPolicy 0 (anonymous) in endpoint 0: configuration doesn't match
	[2020-10-07 12:45:55.999 (UTC+0550)] info/client        Rejecting UserTokenPolicy 1 (certificate) in endpoint 0: configuration doesn't match
	[2020-10-07 12:45:55.999 (UTC+0550)] info/client        Rejecting UserTokenPolicy 2 (certificate) in endpoint 0: configuration doesn't match
	[2020-10-07 12:45:55.999 (UTC+0550)] info/client        Selected Endpoint opc.tcp://DESKTOP-UN398U4:4840 with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
	[2020-10-07 12:45:56.001 (UTC+0550)] info/client        Selected UserTokenPolicy username_basic128 with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
	[2020-10-07 12:45:56.003 (UTC+0550)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
	[2020-10-07 12:45:56.487 (UTC+0550)] info/channel       Connection 268 | SecureChannel 6 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 and a revised lifetime of 600.00s
	[2020-10-07 12:45:56.489 (UTC+0550)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
	[2020-10-07 12:45:56.534 (UTC+0550)] info/client        The ServiceResult has the StatusCode BadCertificateUriInvalid
	[2020-10-07 12:45:56.534 (UTC+0550)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: BadCertificateUriInvalid
	[2020-10-07 12:45:56.537 (UTC+0550)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadCertificateUriInvalid

closed 2020-10-08 by Yashvanthb13

EugenM-sie, 2020-10-23:
@Yashvanthb13 hallo, do you find solution for your problem. because i have the same error

Yashvanthb13, 2020-10-23:
@EugenM-sie  yes i found the solution but i have some new problems now when connecting with server with user name and password i have created new post #3951, can you go through this once and can you suggest me on this what went wrong

EugenM-sie, 2020-10-23:
@Yashvanthb13 i want to ask, how you soved it. I can not genarate in cmake with DUA_ENABLE_ENCRYPTION=ON. I have Generated the certs using create_self-signed.py in tools/cets, but i get the same error

EugenM-sie, 2020-10-23:
@jpfr Hallo, I try to generate code with Cmake gui and with UA_ENABLE_ENCRYPTION=ON. By using create_self-signed.py in tools/cets, I generated two certification(server_cert.der and server_key.der), but in Cmake i get the error:

CMake Error at C:/Program Files/CMake/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:218 (message):
  Could NOT find MbedTLS (missing: MBEDTLS_INCLUDE_DIRS MBEDTLS_LIBRARY
  MBEDX509_LIBRARY MBEDCRYPTO_LIBRARY)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:577 (_FPHSA_FAILURE_MESSAGE)
  tools/cmake/FindMbedTLS.cmake:35 (find_package_handle_standard_args)
  CMakeLists.txt:484 (find_package)

How can i solve it?

Yashvanthb13, 2020-10-23:
you need to include MbedTLS  library location manually and download the latest files

EvgKlin, 2021-06-25:
@juangburgos Hallo, can you maybe help, I start my Server  ./encryptTest.out server_cert.der server_key uaexpert_key.pem
But i get the segmentation fault. Do you, what can be an error

juangburgos, 2021-06-25:
@EvgKlin , seg fault means your are accessing memory that you should not. So probably an error on how you are using pointers. Your question is more specific to programming in C, than to the open62541 library, so sadly I cannot help you with your programming skills. If you have a more open62541 specific question, I am happy to help.

EvgKlin, 2021-06-25:
@juangburgos hallo, i only used Example examples/encryption/server_encryption.c, that's why i was not sure, if i do all right.  I do not know, where i should take certificat for trusted list for server. and i recognize, that python script edit not all rigth, especcialy if i used it on other system.