open62541/open62541 #3927: NodeSet compiler not working when trying to add custom DataTypes

Fischer-Thomas, 2020-10-02:
## Description

When trying to generate code for a custom nodeset + datatypes there are some major problems.
For the start this is my snippet from CMake.
```
ua_generate_nodeset_and_datatypes(
    NAME "custom_A"
    FILE_CSV "${FILE_CSV_DIRPREFIX}/custom/custom_A.csv"
    FILE_BSD "${FILE_BSD_DIRPREFIX}/custom/custom_A.bsd"
    NAMESPACE_MAP "2:http://yourorganisation.org/custom_A/"
    FILE_NS "${FILE_NS_DIRPREFIX}/custom/custom_A.xml"
    INTERNAL
)

add_example(server_nodeset server_nodeset.c
            ${UA_NODESET_CUSTOM_A_SOURCES}
            ${UA_TYPES_CUSTOM_A_SOURCES})

add_dependencies(server_nodeset open62541-generator-ns-custom_A)

``` 
content of my .bsd :
```
<opc:TypeDictionary xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:tns="http://yourorganisation.org/custom_A/" DefaultByteOrder="LittleEndian" xmlns:opc="http://opcfoundation.org/BinarySchema/" xmlns:ua="http://opcfoundation.org/UA/" TargetNamespace="http://yourorganisation.org/custom_A/">
    <opc:Import Namespace="http://opcfoundation.org/UA/"/>
    <opc:StructuredType BaseType="ua:ExtensionObject" Name="MyCustomType">
        <opc:Field TypeName="ua:DataSetMetaDataType" Name="dataSetMetaDataMember"/>
    </opc:StructuredType>
</opc:TypeDictionary>
```
when trying to compile it i get the following error message: 

> RuntimeError: Infinite loop detected or type not found while processing types MyCustomType: unknonwn subtype ['ua:DataSetMetaDataType']. If the unknown subtype is 'Bit', then maybe a struct with optional fields is defined wrong in the .bsd-file. If not, maybe you need to import additional types with the --import flag. E.g. '--import==UA_TYPES#/path/to/deps/ua-nodeset/Schema/Opc.Ua.Types.bsd'

So i didn't quite understand why i need to import a bsd as the Server should already know the DataType "DataSetMetaDataType"(see enabled CMake Flags below).

However like suggested in the error message I then tried it again importing the .bsd like this:

```
ua_generate_nodeset_and_datatypes(
    NAME "custom_A"
    FILE_CSV "${FILE_CSV_DIRPREFIX}/custom/custom_A.csv"
    FILE_BSD "${FILE_BSD_DIRPREFIX}/custom/custom_A.bsd"
    NAMESPACE_MAP "2:http://yourorganisation.org/custom_A/"
    FILE_NS "${FILE_NS_DIRPREFIX}/custom/custom_A.xml"
    IMPORT_BSD "UA_TYPES#${CMAKE_SOURCE_DIR}/tools/schema/Opc.Ua.Types.bsd"
    INTERNAL
)
```
This doesn't work aswell because now it generates the file "types_custom_A_generated.h" which declares all defines from the imported bsd again! (Example compiler Error:
```
In file included from /home/user/repo/open62541/build/src_generated/open62541/namespace_custom_A_generated.h:41:0,
                 from /home/user/repo/open62541/examples/nodeset/server_nodeset.c:13:
/home/user/repo/open62541/build/src_generated/open62541/types_custom_A_generated.h:4607:3: error: conflicting types for UA_PublishedEventsDataType
 } UA_PublishedEventsDataType;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/user/repo/open62541/include/open62541/util.h:12:0,
                 from /home/user/repo/open62541/include/open62541/server.h:19,
                 from /home/user/repo/open62541/examples/nodeset/server_nodeset.c:5:
/home/user/repo/open62541/build/src_generated/open62541/types_generated.h:4756:3: note: previous declaration of UA_PublishedEventsDataType was here
 } UA_PublishedEventsDataType;
```

@Pro @andreasebner

## Background Information / Reproduction Steps

Used CMake options:

UA_ENABLE_PUBSUB=On
UA_ENABLE_PUBSUB_INFORMATIONMODEL=On
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS=On
UA_NAMESPACE_ZERO=FULL

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [x] Operating system: ubuntu 18.04

Pro, 2020-10-02:
Seems this issue was introduced with PR #3922 

I'll check it!

Pro, 2020-10-02:
Will be fixed with #3928.
There was a line which shouldn't be there.

Can you please test with this change:
https://github.com/open62541/open62541/commit/d8bf18208540879b584992c80511492b59d57a7d

Thanks for reporting.

Fischer-Thomas, 2020-10-02:
Removing the line works for me! Thank you for the quick response!

I have one more thing to add:

https://github.com/open62541/open62541/blob/8103c27992a9b2af62c3c607b0cadcf9279433ff/tools/cmake/macros_public.cmake#L451-L452

According to the documentation it should be possible to import multiple BSD however it didn't seem to work for me.
When looking inside of the macro i noticed that "IMPORT_BSD" is handled as a "one value argument" instead of multiple. 

So i moved IMPORT_BSD to line 452 and tried to generate a DataType from my "custom_B.bsd" which has some references from "custom_A.bsd" which required me to import Opc.Ua.Types.bsd (as explained above).
CMake snippet:
```
ua_generate_nodeset_and_datatypes(
    NAME "custom_B"
    FILE_CSV "${FILE_CSV_DIRPREFIX}/custom/custom_B.csv"
    FILE_BSD "${FILE_BSD_DIRPREFIX}/custom/custom_B.bsd"
    NAMESPACE_MAP "2:http://yourorganisation.org/custom_A/" "3:http://yourorganisation.org/custom_B/"
    FILE_NS "${FILE_NS_DIRPREFIX}/custom/custom_B.xml"
    IMPORT_BSD "UA_TYPES#${CMAKE_SOURCE_DIR}/tools/schema/Opc.Ua.Types.bsd"
    IMPORT_BSD "UA_TYPES_CUSTOM_A#${CMAKE_SOURCE_DIR}/examples/nodeset/custom/custom_A.bsd"
    DEPENDS "custom_A"
    INTERNAL
)
```

Generating the .c and .h files worked but building will get an error.
The compiler says in the generated file **types_custom_B_generated.h** it doesn't know the DataTypes generated with "custom_A.bsd" which are located in **types_custom_A_generated.h**. 
Long story short:
After generating the DataTypes i am missing an #include "types_custom_A_generated.h" in **types_custom_B_generated.h**.

Pro, 2020-10-02:
> So i moved IMPORT_BSD to line 452 and tried to generate a DataType from my "custom_B.bsd" which has some references from "custom_A.bsd" which required me to import Opc.Ua.Types.bsd (as explained above).

Could you please create a PR which fixes this. It's already correctly handled here:
https://github.com/open62541/open62541/blob/8103c27992a9b2af62c3c607b0cadcf9279433ff/tools/cmake/macros_public.cmake#L113

> After generating the DataTypes i am missing an #include "types_custom_A_generated.h" in types_custom_B_generated.h.

This may be more complicated to achieve as one generation does not know the resulting file of the other. Maybe it is fixed if you include the corresponding header files in the right order in the main C file?

commit d8bf1820 (2020-10-02): fix(datatype-generator): Correctly handle existing BSD definitions. Fixes #3927
commit aa089959 (2020-10-02): fix(datatype-generator): Correctly handle existing BSD definitions. Fixes #3927
closed 2020-10-02 by Pro