open62541/open62541 #3931: Generate notifications with same value rewrite in monitoredItems.

jugurthab, 2020-10-04:
Hi,

I'm new to OPC UA and open62541.

I'd like to know if there is a way in open62541 monitoredItems (https://open62541.org/doc/current/tutorial_server_monitoreditems.html) to generate a notification when the value has been re-written (even if it has been re-written with the same value)?

Thanks.

basyskom-meerkoetter, 2020-10-07:
Yes you can.

OPC UA allows you to create monitored items that trigger either on value change, status chances or a changed datasource timestamp.

https://reference.opcfoundation.org/v104/Core/docs/Part4/7.17.2/

So on the client side you need to create a monitored item which instructs the server to trigger on datasource timestamp changes (in addition to value changes). On the server you need to make sure to update the data source timestamp on each read. See https://open62541.org/doc/current/tutorial_server_datasource.html

basyskom-meerkoetter, 2020-10-08:
Thinking about it - why use a MonitoredItem in the first place? This sounds more like you like to read the variable over and over again (client side polling). Note that OPC UA allows to read several values via a single request.

jpfr, 2020-10-20:
Good answers. Please reopen if necessary.

closed 2020-10-20 by jpfr