open62541/open62541 #3941: undefined reference to UA_Server_createEvent and UA_Server_triggerEvent

abdo4694, 2020-10-07:
## Description

I am trying to get the library into two files open62541.c and open62541.h. by activating the UA_ENABLE_AMALGAMATION. 

## Background Information / Reproduction Steps

Everything is working fine with the simple server example. However, I am trying to implement the following example in the source file: tutorial_server_events.c in the Examples folder. 
I get the following errors: 
undefined reference to 'UA_Server_createEvent'
undefined reference to 'UA_Server_triggerEvent'
I can find the declaration in the header file open62541.h but not the implementation in the source file open62541.c.

Used CMake options:

[image]

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): Clone master branch on 22 Sep
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

juangburgos, 2020-10-07:
Have you tried enabling `UA_ENABLE_SUBSCRIPTIONS_EVENTS=ON` and `UA_NAMESPACE_ZERO=FULL` ?

abdo4694, 2020-10-07:
@juangburgos On this link https://open62541.org/doc/current/building.html
It says that UA_ENABLE_SUBSCRIPTIONS_EVENTS is experimental, is it ok to make it on?

juangburgos, 2020-10-07:
I have been using it for a while in my [QUaServer project](https://github.com/juangburgos/QUaServer) without any issues :information_desk_person:

abdo4694, 2020-10-07:
Oh ok great thank you
but when I try to make UA_NAMESPACE_ZERO=FULL
I get the following
CMake Error at CMakeLists.txt:1005 
../open62541/deps/ua-nodeset/Scheme/Opc.Ua.NodeSet2.xml not found. You probably need to initialize the git submodule for deps/ua-nodeset
Do you have any idea about that?
Thank you in advance :)

juangburgos, 2020-10-07:
Yeah, you need to clone open62541 **recursivelly**:

```
git clone --recursive https://github.com/open62541/open62541.git open62541.git
```

Or if already cloned, update submodules recursivelly

```
cd open62541.git
git submodule update --init --recursive
```

abdo4694, 2020-10-07:
@juangburgos It worked ! Thanks a lot for your help :)

juangburgos, 2020-10-08:
No prob, you can close this issue if there is no more to it

closed 2020-10-08 by abdo4694